MarlinFirmware/Marlin #9135: [FR] Anet A2 Support ?

dustinreeves, 2018-01-10:
https://github.com/thijsk/Skynet3d/tree/master/SkynetV2.3.2/Configuration%20Profiles/Anet%20A2

^-Can we port these anet a2 configuration profiles into marlin current ?

fiveangle, 2018-01-10:
Yes, it should not be hard.  Easiest is to use some text comparison utility such as WinMerge, BeyondCompare, or even "_vim -d_" then migrate the settings by hand to the new Config*

-=dave

dembach, 2018-01-10:
maybe it can be moved to the example folders and altered to the new marlin version..

dustinreeves, 2018-01-10:
I'm half ass handy I'll do a diff and see if I can do my first merge tonight (was really hoping someone had already done the legwork for me :p )

dustinreeves, 2018-01-14:
this has turned into a much more daunting task than i originally imagined, (very new to 3d printing)  Will probably wait this out until someone with more experience chimes in ?

diegohsoriano, 2018-01-30:
Same case here, I need to install Marlin into an Anet A2.  Could you please share the instructions to create the config files? Thanks! :)

thinkyhead, 2018-02-14:
Keep on trying. Examine the example configurations for guidance. Get a "merge" or "diff" plugin for your favorite text editor. Compare the default configs to some of the example configs for practice. You'll get it.

IngoTheMan, 2018-03-22:
Is it only the configuration.h and configuration_adv.h files that need to be changed to make this work? I did take a quick look at the files and it seems that the configuration_adv.h file don't need to be changed at all. Also, if I manage to port skynet 2.3.2 settings to newest marlin (I'm totally new at this) how do I share my work?

thinkyhead, 2018-03-29:
@IngoTheMan — http://marlinfw.org/docs/development/getting_started_pull_requests.html

beukhof1, 2018-07-09:
@IngoTheMan Can u please share your config? And is it working right?

nicopenna, 2018-08-06:
Hi @beukhof1 did you tried the firmware?? I getting some problems with marlin 1.1.9 and my anet A2.