MarlinFirmware/Marlin #9136: pause/stop print will resume when different print is startred

v1ru3, 2018-01-10:
- Description: If during a print pause is invoked and then stop print, next time when a different print is started (completely different file from the SD card is chosen), the printer tries to resume print instead of printing the new file.

- Steps to reproduce:
1. start a print
2. pause
3. abort
4. start a different print.

If power is cycled between 3 and 4, expected behavior is normal.

on 1.1.8 and latest bugfix.

Nand-e, 2018-01-11:
I experienced the same. And report https://github.com/MarlinFirmware/Marlin/issues/9063

But it is a hard bug, not a configuration issue.

v1ru3, 2018-01-11:
Thanks i wasn't aware about this being already open. I know it's a bug. I guess it can be closed as duplicate.

Quick question? Have you used pause/resume? When i use it, i get a gap in the layers exactly where pause/resume is used. Not sure if it's related to the extruder or resuming too high.

v1ru3, 2018-01-13:
Same, printer, extruder... etc. I used pause in 1.1.5 and i don't remember any gaps between layers.

closed 2018-02-11 by thinkyhead

v1ru3, 2018-02-11:
this is exactly the same with the latest bugfix i downloaded tis morning....

thinkyhead, 2018-02-14:
Let's continue at #9063.