MarlinFirmware/Marlin #9138: [FR][bugfix_2.0.x] Pull Request for Delta AutoCalibration Algorithm of Minor Squares based on DC42

dembach, 2018-01-10:
Implement Delta AutoCalibration Algorithm of Minor Squares based on DC42 as alternative and fast way to calibrate a Delta like David Crocker 

/**
   * Delta AutoCalibration Algorithm of Minor Squares based on DC42 RepRapFirmware 7 points
   * Usage:
   *    G33 <Sn> <Pn> <Q>
   *      S = Num Factors 3 or 4 or 6 or 7
   *        The input vector contains the following parameters in this order:
   *          X, Y and Z endstop adjustments
   *          Delta radius
   *          X tower position adjustment and Y tower position adjustment
   *          Diagonal rod length adjustment
   *      P = Num probe points 7 or 10
S=-1 Don't adjust anything, just print the height error at each probe point

S=0 Equivalent to S=<number_of_points_probed>

S=3 Adjust homing switch corrections only

S=4 Adjust homing switch corrections and delta radius

S=6 Adjust homing switch corrections, delta radius, and X and Y tower position offsets

S=7 Adjust homing switch corrections, delta radius, X and Y tower position offsets, and diagonal rod length

The adjustments are made so as to minimise the sum of the squares of the height errors.

See Also:
[https://duet3d.com/wiki/Configuring_RepRapFirmware_for_a_Delta_printer](url)
[https://github.com/dc42/RepRapFirmware](url)
[https://github.com/MKFirmware/MK4duo/blob/master/MK4duo/src/core/commands/gcode/delta/g33_type1.h](url)

thinkyhead, 2018-02-11:
@LVD-AC — What do you think about this concept?

LVD-AC, 2018-02-11:
DC42 and LVD are based on the same math model called displacement method:
1) construct a displacement matrix
2) invert it
3) iterate though the inverted matrix until a solution is found.

The differences are:
- in LVD G33 step 1 and 2 are done once and for all in an excel sheet for a circular grid; the inverted matrix only needs to be scaled to the printer dimensions. DC42 builds the matrix from scratch every time and as such requires forward delta kinematics to be implemented and more memory
- DC42 is limited to the number of points (internal memory) but is more versatile one can choose any points; LVD imposes a grid of points to be used (up to to 100 points)
- LVD probes and iterates automatically, DC2 needs for the user to probe a grid and reissue G33 until satisfied - this non-automated approach is probably due to the diagonal rod calibration included in DC42 but which is left out in LVD G33 because it is very unstable and does not iterate properly.

Both methods are as fast and as accurate and lead to the same results especially after #8822 which calculates the required scaling factors with (approximate) forward kinematics.

If someone is up to it we can have 2 flavors of G33 like in Marlin-Kimbra, similar to having different G29 flavors.

dembach, 2018-02-15:
@LVD-AC : 
nice explanation.
so... both ways are good. DC42 is very fast and is doing all in one. but the delta kinematics vary from calibration to calibration a little bit, i can confirm that...

LVD needs more time due to more points, but the kinematics are not set up automatically as far as i understand.

is there a chance to get the best of both ? Is there some algorithm to do the diagonal rod calibration in a matter that is more stable ? 

if there would be a implementation of both flavors like in Marlin-Kimbra, there should be a good and easy description , what the user may expect for each Flavor. 
Since we are talking about newer architectures of CPU´s with more RAM, it should be possible to expand capabilities of DC42´s approach to have more points and more accuracy. Right ?

Sorry for the curious questions ... Just trying to understand and trying to get deltas , as mine, to work better and just perfect. i love watching the moves of my delta.

LVD-AC, 2018-02-15:
> but the kinematics are not set up automatically

What do you mean by that?

dembach, 2018-02-15:
diagonal rod calibration

LVD-AC, 2018-02-16:
On an entry-level starter kit you can achieve much more precision in diagonal_rod calibration with a calipers than you can achieve with Z-probing.