MarlinFirmware/Marlin #9139: [FR] - review and simplify  the whole logic of endstops (Auto Home) and related "parameters"

novreis, 2018-01-10:
Assumptions and Background
1) I fully appreciate the work of all  contributors of Marlin so no offense with this comment
2) my professional career was mainly as CIO
3) I downloaded a version of Marlin in theory adapted to Tronxy printers as mine 
4) I am a newbie at 3d printing
5) M119 is working correctly when I press each interrupt
After experiencing the problem that a lot of Marlin users also had - endstops are not "respected"  at Auto Home - I googled a lot about this issue and the solutions are not satisfactory. 
a) I cannot understand any logic that justifies that after a end stop is triggered the motor tries to go past the end stop! Neither inconsistencies inside my downloaded firmware!
more ...
b) if the printer is printing correctly how can it fail at a simple Auto Home? 
Some features must override all other, in this case Auto Home must override everything like "directions", "inversions", "endstops_only_for _homing" and so on.
If Auto Home is triggered, Marlin must "understand"  where the hot end was before going "home". Knowing this, when the end stop is triggered no command can be issued that take the hot end beyond this limit! Sorry to say but It seems simple for me!
Sorry for my English and be opened to review the logic, at least of Auto Home!
Go on with your excellent work!

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

AnHardt, 2018-01-10:
> If Auto Home is triggered, Marlin must "understand" where the hot end was before going "home". 

If we'd know where we are, we'd have no need for homing.
Homing is no 'Go To'. It's 'find out where you are'.

novreis, 2018-01-11:
Ok, I  can understand that after a manual movement of the hot end, Marlin does not know where it is, but after a Print the last command that position the hot end (x,y,z) is known, I assume!
Anyway, I must make further tests to be sure of what I was saying about a second option to cope with this issue. Thanks for your answer!

lrpirlet, 2018-01-11:
@novreis  

> _M119 is working correctly when I press each interrupt_

This seems to mean that you did configure the endstops to interrupt when triggered... Try using the polling method in the configuration.h file.
 (do comment back this line ` #define ENDSTOP_INTERRUPTS_FEATURE` to read ` //#define ENDSTOP_INTERRUPTS_FEATURE` )

If that solve your problem, please upload the configuration.h (where I could have read the variable) and the marlin version used... A long time ago, I jumped back from interrupt to polling because of intermittent hot end crash landing in the bed... we may still see a bug.

novreis, 2018-01-11:
Before wasting your time I must try to understand the randomness of my issue.
At first was the y end stop that was failing. Yesterday both x and y end stops were failing but today I noticed that one wire was broken at x end stop. After solder the wire no more fails at x e y!!!! I did not make any intervention at y end stop so my surprise today! But before, my Auto Home was making the bed reaching the y end stop. Now is not going back!!! As I said, I need further tests to confirm these things. Meanwhile I am not changing any parameters related with this because I think they are confusing for a noob! 
About //#define ENDSTOP_INTERRUPTS_FEATURE  mine is commented like this!
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1623659/Configuration.txt)
Thks

closed 2018-02-11 by thinkyhead