MarlinFirmware/Marlin #9140: [Question] 1.1.8 TMC2130 initialization

ABH10, 2018-01-10:
I am debugging why my TMC2130 doesn't work properly. When I measure the SPI datagrams written to TMC2130 during initialization, there seems to be missing a lot. Apparently it is so that each function in the registers are added sequentially and there are writes to TMC2130 for each function. I.e. instead of just having one write to a register that includes all functions, we have maybe 8 writes with different number of defined functions, before we get to the final write, that includes all functions! While this should work, I however experience that not very many datagrams are actually present on the SPI bus.
The very first write on the SPI bus goes to the GCONF register with data content 0x00000000. I.e. everything is cleared in GCONF. There are no later writes to GCONF:
[image]

For each TMC2130 I can only see 7 writes (and no reads) in total. These are as follows:

MOSI                MISO
80 00 00 00 00      00 00 00 00 01      Write GCONF = 0, Stealthchop = 0
EC 00 00 00 00      39 00 00 00 00      Write CHOPCONF = 0
ED 00 00 00 00      39 00 00 00 00      Write COOLCONF = 0      
F0 00 05 04 80      39 00 00 00 00      Write PWMCONF = reset default
90 00 00 00 00      39 00 05 04 80      Write IHOLD_IRUN = 0
EC 00 00 00 08      39 00 00 00 00      Write CHOPCONF, TOFF = 8
EC 00 00 80 08      39 00 00 00 08      Write CHOPCONF, TOFF = 8, TBL = 1 (24 clocks)

EC 00 02 80 08      39 00 00 80 08      Write CHOPCONF, TOFF = 8, TBL = 1 (24 clocks), vsense = high
90 00 00 19 00      39 00 02 80 08      Write IHOLD_IRUN, IRUN = 25
90 00 00 19 0C      39 00 00 19 00      Write IHOLD_IRUN, IRUN = 25, IHOLD = 12

Clearly a lot is missing. For example I have STEALTHCHOP defined, but "en_pwm_mode" in GCONF is obviously set to 0.
The physical interface seems to work perfectly well. All writings are replayed on the MISO line in the following writes. This proves that TMC2130  received the datagrams.
I have stepper RUN current set to 800 mA. That corresponds to IRUN = 25 and vsense = high, so something is set correct.

Any idea why this is not working for me?
I use Marlin 1.1.8 release and have TMC2130Stepper version 2.0.2 installed.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1620680/Configuration.zip)

teemuatlut, 2018-01-11:
The commands are given sequentally for easier reading and debugging. You're right that the registers could be formed first and then pushed to the driver. I might do switch back to that at some point but this has worked well enough for now.

You can see the register clearing commands in `TMC2130Stepper::begin()`. This clears registers GCONF, CHOPCONF, COOLCONF, PWMCONF and IHOLD_IRUN. Then it sets the default off time and blank time values.
After begin, Marlin will start configuring the driver and first sets the current through `TMC2130Stepper::setCurrent()`. This will set the VSENSE bit in register and then set the IRUN and IHOLD values.

I believe this is as far as you could see with your analyzer.
The next step would be to set the microsteps in CHOPCONF.
But could it be that with your analyzer settings, this is the point it runs out of memory and cannot log values any longer?

Btw, what analyzer do you have? I just ordered a cheap one myself for this very same purpose!

ABH10, 2018-01-11:
I did monitor for quite long time after the sequence above, but I will take another look later today.
I believe the next writing after the sequence above should be like this:

EC 00 02 80 08 39 04 00 80 08 Write CHOPCONF, TOFF = 8, TBL = 1, vsense = high, MRES = 4

where microstepping is being set to 16.
I will try to set up a trigger specifically on this SPI writing.

I am using Rigol DS1054Z with all options added (corresponds to DS1104Z specs.). I believe DS1054Z is discontinued and on sale right now.

ABH10, 2018-01-11:
I found the missing commands.
For some reason, in my system there is a delay of approx. 320 ms between
these two function calls:

st.setCurrent(st.getCurrent(), R_SENSE, HOLD_MULTIPLIER);
.. 320 ms
st.microsteps(microsteps);

I logged the remaining communication and it was OK.

I also found the issue I had with the SPI:
The pins I have used for CS were also used for a display (that I don't use).
This display was initialized after the stepper initialization and the CS pins would then be redefined for display use.
I removed this display and now the SPI works all the time.

I would like to propose that an SPI write check is added to Marlin.
For each write you could test that the command was read and understood by TMC2130 by comparing the replay on the MISO line for the next operation with the written datagram.
If there is a mismatch it should trigger an error message.

closed 2018-02-11 by thinkyhead