MarlinFirmware/Marlin #9142: Trying to understand LIN_ADVANCE...again :-)

VanessaE, 2018-01-11:
This is Marlin 1.1.6 (commit  8dc10fb4c206e4bbcbe9ea21a25d5a9802aa54a8)

So...I decided to give LIN_ADVANCE another go, since I have since upgraded to a newer Marlin, a new hotend (now a genuine J-head v8), a new geared direct-feed extruder of my own design (https://www.thingiverse.com/thing:2744682).  Instead of wasting a bunch of filament on random models, I went for the LIN_ADVANCE calibration test pattern (http://marlinfw.org/tools/lin_advance/k-factor.html)... and rather than ask a bunch of questions, maybe I should let some photos do the talking:

K=50 to 150, speeds of 5 mm/s and 100 mm/s (to test a 20:1 flow rate difference, similar to kinds of differences Slic3r may produce during a typical print, say an interior perimeter or sparse infill versus the thinnest bits of gap fill).

[image]

K=50 to 150, speeds of 40 mm/s and 80 mm/s (to test a 2:1 flow rate difference)

[image]

When the difference is small, the test pattern can get pretty close to perfect.  Indeed, previous tests showed K=42 to be the sweet spot for the test pattern, but that setting is quite inadequate when I try a real calibration print e.g. a Benchy or Ctrl-v v3.

In Marlin, I've left the E:D ratio at 0/auto.

What do I do now?

Here's the gcode output from the 40/80 test (the 5/100 test differed only by speed, all other parameters unchanged):
```gcode
; ### Marlin K-Factor Calibration Pattern ###
; -------------------------------------------
;
; Created: Wed Jan 10 2018 21:51:52 GMT-0500 (EST)
; Settings:
; Print Size X = 170 mm
; Print Size Y = 125 mm
; Print Rotation = 0 degree
; Print Pattern = Standard
; Print Frame = true
; Prime Nozzle = true
; Prime Extrusion Multiplier = 2.5
; Dwell Time = 2 s
; Filament Diameter = 1.72 mm
; Nozzle Diameter = 0.4 mm
; Nozzle Temperature = 215 °C
; Nozzle / Line Ratio = 1
; Bed Temperature = 60 °C
; Slow Printing Speed = 1200 mm/min
; Fast Printing Speed = 6000 mm/min
; Printing Acceleration = 2500 mm/s^2
; Movement Speed = 7200 mm/min
; Use BL = true
; Retraction Distance = 2.5 mm
; Bed Size X = 200 mm
; Bed Size Y = 200 mm
; Origin Bed Center = false
; Layer Height = 0.208 mm
; Extrusion Multiplier = 1
; Starting Value K-Factor = 50
; Ending value K-Factor = 150
; K-Factor Stepping = 5
; Test Line Spacing = 5 mm
; Test Line Length Slow = 40 mm
; Test Line Length Fast = 80 mm
;
; prepare printing
;
G28 ; home all axes
M190 S60 ; set and wait for bed temp
M104 S215 ; set nozzle temp and continue
G29 ; execute bed automatic leveling compensation
M109 S215 ; block waiting for nozzle temp
G21 ; set units to millimeters
M204 P2500 ; set acceleration
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
G92 E0 ; reset extruder distance
;
; prime nozzle
[... 552 lines cut ...]
  */
-//#define LIN_ADVANCE
+#define LIN_ADVANCE
 
 #if ENABLED(LIN_ADVANCE)
-  #define LIN_ADVANCE_K 75
+  #define LIN_ADVANCE_K 42
 
   /**
    * Some Slicers produce Gcode with randomly jumping extrusion widths occasionally.
diff --git a/Marlin/Marlin.ino b/Marlin/Marlin.ino
index 49b6fd14b..dd7b7c966 100644
--- a/Marlin/Marlin.ino
+++ b/Marlin/Marlin.ino
@@ -1,5 +1,6 @@
-/**
+ /**
  * Marlin 3D Printer Firmware
+ * 1.1.6 release, commit 8dc10fb4
  * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
  *
  * Based on Sprinter and grbl.
-- 
2.11.0
```

Sebastianv650, 2018-01-12:
Please give us an indication of which line is first and which one is the last one and also of the print direction of the lines. I'm a little bit confused at the moment.

VanessaE, 2018-01-12:
The lines closest to the camera are first, and all were printed left-to-right.

Sebastianv650, 2018-01-13:
I guess your extruder is skipping steps during the high K lines, at least that's the only explanation I have for seeing the opposite compared to what I expect from increasing K.
Other things I don't feel comfortable with:
- Your high speed test with 100mm/s results in a flow rate of 10mm³/s. That's realy a lot and I know from my own hotend it's the absolute upper limit where I can get a stable extrusion flow. Have you tested this flow rate without LIN_ADVANCE? I recommend to print a very big cylinder in vase mode for testing while feed rate is increased until the wall isn't clean any more.
- Your acceleration is set to 2500mm/s². At least for the test pattern I would recommend a much lower value like 500 as it eleminates other possible issues and it's much more easy to see whats happening during acceleration because now there is a reasonable acceleration distance.

I recommend printing another test pattern, but this time stick to the default values in the Speed and K-factor sections.

Sebastianv650, 2018-01-13:
Just for information, I'm working at an improved version 2 of LIN_ADVANCE at the moment. You can follow the status (even if I don't write every progress there) in #9048.
It will take some time as it will contain a more "intelligence" than v1 and I was sorting out some odd planner behaviour first. After a PR the planner looks good now so I continue with v2 development again.

VanessaE, 2018-01-13:
> I guess your extruder is skipping steps

I doubt it. 47:9 gearing, a brand new hob gear (less than 2 weeks old, not even 1 kg of filament put through it), and 100% certainty that the hotend is clean, and a spring/idler capable of crushing the filament flat (though obviously I don't have it tensioned to do that :smile:).  At worst, the hob might slip on the filament, but this extruder design makes a distinct sound when it does.

> Your high speed test with 100mm/s results in a flow rate of 10mm³/s. That's realy a lot and I know from my own hotend it's the absolute upper limit where I can get a stable extrusion flow.

For a genuine J-head, 10mm³/s is pretty low.  According to the manufacturer, it's capable of about twice that flow rate; indeed I've pushed it to 200 mm/s (0.4 mm nozzle, 0.304 mm layer height) and while those speeds make my bot's other mechanical inadequacies apparent, the extruder/hotend performed as designed - of course at the time, I was not using LIN_ADVANCE.  But I get your point.

> Your acceleration is set to 2500mm/s². At least for the test pattern I would recommend a much lower value like 500 

OK, I'll give that a try.

Sebastianv650, 2018-01-14:
> I doubt it. 47:9 gearing, a brand ..

It's not a question of quality but of inertia. The stepper itself has to accelerate to execute the step signal. If it gets too much steps within a too short amount of time, it will just not follow the signal and jump to the next stable full-step position.

VanessaE, 2018-01-14:
Oh right, makes sense.

closed 2018-02-11 by VanessaE