MarlinFirmware/Marlin #9145: Z axis software endstop doesn't work

dphamminh, 2018-01-11:
### Bug Report
- Description: 
      Geeetech Prusa Aluminium I3 with sanguinololu board ATMega1284 16Mhz
      I use Repetier Host to manually control the movement on X, Y and Z axis.

- Expected behaviour:
when I enter a displacement above the MAX_POS, I expect printer to be blocked at MAX_POS
 
- Actual behaviour: 
Functioning is correct on X and Y axis, but there is no control on Z axix

- Steps to reproduce:
  - Set Z to home position
  - Increase Z position above max position using repetier host (eg 210mm)

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1621576/Configuration.zip)

dphamminh, 2018-01-11:
I also noticed that when I ask for 10 mm up movement, the Z axis moves 6.5 cm up. So may be there is something wrong in the configuration file. The lead screw is 8mm.

fiveangle, 2018-01-11:
It's obvious you just renamed the version number in your very old config and crossed your fingers that it would work.  Go back and do the work to hand-port your configuration to the new one and you'll surely find where you messed up.

-=dave

dphamminh, 2018-01-12:
I checked my configuration.h file and found that the setting of the Z axis was incorrect. The Z axis software endstop works perfectly now. 

Sorry for the inconvenience and many thanks for your help and advice. I will make a new configuration.h file using the newest templates.

closed 2018-02-11 by thinkyhead