MarlinFirmware/Marlin #9151: Bug Report: #define AUTO_BED_LEVELING_BILINEAR not Triggering M600 on Filament Runout

dcwalmsley, 2018-01-11:
### Bug Report
- Description: #define AUTO_BED_LEVELING_BILINEAR not Triggering M600 on Filament Runout.

- Expected behavior: When Dyze Design's Filament Sentinel sensors detect no filament, the execution of M600 should occur and await operator to feed new filament into sensor/bowden.

- Actual behavior: Nothing happens when filament is removed. However, if I recompile and flash controller with #define AUTO_BED_LEVELING_UBL enabled, I do see M600 being executed and extruder parked with LCD alert to feed filament prior to resuming.  

- Steps to reproduce:  Tested with success on all four filament sensors daisy chained.

Suspect #define AUTO_BED_LEVELING_BILINEAR is broke.

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

r,
Doug Walmsley

dcwalmsley, 2018-01-11:
[Marlin 1.1.6.zip](https://github.com/MarlinFirmware/Marlin/files/1624402/Marlin.1.1.6.zip)

dcwalmsley, 2018-01-12:
New issue though.  Seems my printer at some stage initiates a M600 code while printing but all filament sensors have filament in them.  Another Bug?

thinkyhead, 2018-02-11:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

boelle, 2019-02-20:
@dcwalmsley Please test with the latest `bugfix-2.0.x`

dcwalmsley, 2019-02-20:
This is old and out of date. Closing.

closed 2019-02-20 by dcwalmsley