MarlinFirmware/Marlin #9154: MarlinFirmware is now on Slack

thinkyhead, 2018-01-12:
https://marlinfirmware.slack.com/

This is a web-based chat style interface with Channels similar to IRC. It adds the ability to embed media including formatted code. This should help to improve the back and forth between hardcore code monkeys where the forum style doesn't do as well. Slack itself is a fairly unobtrusive service, and sign-up takes seconds.

I invite the @MarlinFirmware/general-maintainers and regular contributors to join.

@tcm0116 @teemuatlut @LVD-AC @akaJes @Sebastianv650 @Bob-the-Kuhn @AnHardt @p3p 

Feel free to invite any others I have omitted.

[image]

teemuatlut, 2018-01-12:
Great, but how do I get on?
> Don't have an account on this workspace yet?
Contact the workspace administrator for an invitation

I created an account but it doesn't work for Marlinfirmware. All I can do with it is create a new workspace.

LVD-AC, 2018-01-12:
I got this:

 

We couldn't find workspaces for this address.

If you're looking for a specific workspace, you can try contacting your
Workspace Administrator for an invitation.

 

I'll just wait for formal r.s.v.p. in the mail.

 

https://marlinfirmware.slack.com/

 
<https://user-images.githubusercontent.com/698003/34865907-282690d2-f740-11e
7-90f7-d74e968930fb.png> image

-
<https://github.com/MarlinFirmware/Marlin/issues/9154> , or mute
lIYVBxHks5tJxm3gaJpZM4Rb96_>  the thread.
<https://github.com/notifications/beacon/AZjUYWTTLeRdte0QAiwQGPfb6ddEH9Fmks5
tJxm3gaJpZM4Rb96_.gif>

psavva, 2018-01-12:
I would approve an invite too @thinkyhead

Bergerac56, 2018-01-12:
I would like to be "invited" too ;)

Sebastianv650, 2018-01-12:
So do we need an invitation before we can register or vice versa?

fiveangle, 2018-01-12:
Most other projects that use slack simply set permissions so anyone can join 🤷‍♂️

Zakm0n, 2018-01-13:
In Replicape's slack they have permissions set where contributors and devs have their own code monkey channel, and there is general troubleshooting and config help separately. I feel like it is much more accessible than using github's issues forum for newer users trying to troubleshoot.

oysteinkrog, 2018-01-13:
For what it's worth I agree with what Zakm0n said.

jbrazio, 2018-01-15:
@thinkyhead I can't join, it tells me to contact you.

LVD-AC, 2018-01-15:
I can created a workspace with my email and could set up a password for it but that login doesn't work.

thinkyhead, 2018-01-15:
Ah, I have to collect email addresses… Best way to find me. Email me through Github maybe?

boelle, 2019-06-22:
should be pinned

closed 2019-07-21 by boelle