MarlinFirmware/Marlin #9155: Marlin 2.0 Build for MKS SBase 1.3

marcalcc, 2018-01-12:
I'm trying to build the firmware.bin file to flash my MSK SBase 1.3 board, following the steps found here: https://github.com/MarlinFirmware/Marlin/issues/8131 (I haven't found any other documentation on the matter).

I just downloaded and changed the Configuration.h and Configuration_adv.h for those found as examples for my board.

No matter what I cannot build, among a lot of verbose output, the console shows an error:

Marlin/src/gcode/eeprom/M500-M504.cpp:74:5: error: no matching function for call to 'MarlinSettings::report(bool, int16_t&)'

Any help would be appreciated.

Thanks in advance

tig33r, 2018-01-12:
Show both configuration files here.

marcalcc, 2018-01-12:
Thanks for your reply,

I'm very new to this environment and I don't know how to show you the files. These are the ones found on

Marlin/src/config/examples/Mks/Sbase

I'm just trying to figure out how to build it and flash it, if I work it up, I will start modifying the files to fit my printer

jameswood, 2018-01-14:
I had the same issue. In configuration.h I changed ```#define SERIAL_PORT 0``` to ```#define SERIAL_PORT -1``` and commented out ```// #define SERIAL_PORT_2 -1```

It seems that the second serial port isn't supported.

marcalcc, 2018-01-14:
Many thanks, @jameswood that was spot on! I'm so much looking forward to try this out

closed 2018-01-14 by marcalcc

thinkyhead, 2018-01-15:
This might also do the trick:
```diff
  void GcodeSuite::M503() {
    (void)settings.report(parser.seen('S') && !parser.value_bool()
-     #if NUM_SERIAL > 1
+     #if ADD_PORT_ARG
        , command_queue_port[cmd_queue_index_r]
      #endif
    );
  }
```

marcalcc, 2018-01-15:
many thanks @thinkyhead  
I see you committed this already, so I assume there is no need to change the configuration.h on new downloads of the project. Thanks again and keep up the awesome work!

marcalcc, 2018-01-15:
**Update**

The build still does not compile unless you do the changes suggested by @jameswood

jameswood, 2018-01-15:
@marcalcc let's call my suggestion what it is, a hacky workaround! The changes made by @thinkyhead  cleared up the original error for me, and now I'm getting the following:

```
Marlin/src/module/configuration_store.cpp:1836:8: error: prototype for 'void MarlinSettings::report(bool, int8_t)' does not match any in class 'MarlinSettings'
void MarlinSettings::report(const bool forReplay
^~~~~~~~~~~~~~
In file included from Marlin/src/module/configuration_store.cpp:47:0:

Marlin/src/module/configuration_store.h:88:19: error: candidate is: static void MarlinSettings::report(bool)
static void report(const bool forReplay=false
^~~~~~
*** [.pioenvs/LPC1768/src/src/module/configuration_store.o] Error 1
```

@thinkyhead - I was getting these errors before my workaround. These errors are not appearing when compiling with the second serial port disabled. *With* my workaround, I was having some strange serial port behaviour (eg, when PWM autotuning, the results would be truncated to the point that Kp wasn't shown) which are possibly related (?), but I could submit an Issue for that as well if it would be useful.

marcalcc, 2018-01-16:
exactly, same here. Thanks

thinkyhead, 2018-01-22:
Applied the patch described above. It should be better now.

Ofr3d, 2019-03-15:
Hey i'm receiving an error msg saying I have to specify 'upload port', got any idea what should i put in?

commit d178d6e9 (2018-01-15): M503 patch for EEPROM_CHITCHAT off

Fix #9155