MarlinFirmware/Marlin #9156: [2.0.x] Unable to save EEPROM on Re-ARM

tcm0116, 2018-01-12:
I'm on https://github.com/MarlinFirmware/Marlin/commit/da600e5fbed07c98580681fb81be0077deda7f56 with a Re-ARM board on my Delta machine. Trying to save the EEPROM results in a size error. I enabled `DEBUG_EEPROM_READWRITE`, and it failed on `home_offset`. I ended up adding `_FIELD_TEST` checks for each of the `planner` values between `esteppers` and `home_offset`, and it failed at `planner_axis_steps_per_mm`. Everything looks correct in `SettingsData` as well as in `HAL::PersistentStore::write_data` for the LPC1768 HAL, so I'm not sure where the issue lies.

p3p, 2018-01-12:
Same issue as mentioned in #9056? Haven't had time to look into it yet was hoping thinkyhead would fix it before I had to ^^

tcm0116, 2018-01-12:
@p3p Yes, it is the same issue. Hopefully the details I provided above will
help with the diagnosis. It seems like there's an issue with eeprom_index,
but it wasn't obvious.

cjsoong, 2018-01-13:
I use arduino due + ramps fd is the same problem.
This problem is added G-code M504 began to appear,
Before 2017-12-05 download version is ok

tcm0116, 2018-01-13:
Turns out that it's due to the variables within structs being aligned on 4-bytes boundaries on 32-bit systems. Just submitted a PR to address it.

Roxy-3D, 2018-01-13:
Good change!   And this is mostly a rhetorical question.   But now would be a good time to say everything in the 32-bit world is aligned on DWORD boundaries if that made sense.   Do we really need to save every byte of data space in the 32-Bit world?

tcm0116, 2018-01-13:
@Roxy-3D for the EEPROM, it needs to be byte aligned since the values are
written without padding. The structure that gets wrapped by the #pragma is
actually only used to determine what the length of the EEPROM data should
be.

closed 2018-01-15 by tcm0116

chrisjenda, 2018-01-19:
I have tried again with the latest #9236 merge and I am still getting the error.   I have the default config consisting of only two changes; the Motherboard Define being set to BOARD_RAMPS_14_RE_ARM_EFB and EEPROM_SETTINGS un-commented.  This causes the following error on M500

Recv: Error:EEPROM datasize error.
Changing monitoring state from 'Operational' to 'Error: EEPROM datasize error.

tcm0116, 2018-01-19:
Can you try enabling `DEBUG_EEPROM_READWRITE` in configuration_store.cpp and report the result?

chrisjenda, 2018-01-19:
Recv: Error:Field lpq_len mismatch.
Changing monitoring state from 'Operational' to 'Error: Field lpq_len mismatch.

tcm0116, 2018-01-19:
I found the problem. Line 550 of configuration_store.cpp needs to change
from int to int16_t for the lcd_preheat variables. Can you try that and
confirm (my printer is currently occupied printing a Bullet Bill).

tcm0116, 2018-01-19:
I just submitted #9240 with the change above. That should resolve the issue for when there's not an LCD.