MarlinFirmware/Marlin #9157: [bugfix-2.0.x][bug] M43 on RAMPS1.4/MEGA2560 not working

dembach, 2018-01-12:
### Bug Report
- Description: M43 Pin Debug kills Arduino. 
- Expected behaviour: output of PINs Status
- Actual behaviour: Arduino killed, no response anymore , fieeping sound (not the buzzer, its electronics)
- Steps to reproduce: execute M43
  - 
  - Hardware BiQu BigTreeTech V1.4 (RAMPS 1.4 on MEGA 2560)

Attached  `Configuration.h` and `Configuration_adv.h`.

[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1627603/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1627604/Configuration.txt)

Bob-the-Kuhn, 2018-01-12:
I wasn't able to reproduce the problem.  

I downloaded bugfix-2.0.x, copied in your config files and downloaded it to my 2560 with RAMPS.  **M43** and **M43 I** both worked as expected.

Any suggestions on how to reproduce the problem?

Bob-the-Kuhn, 2018-01-13:
The above was with PlatformIO.

When I used Arduino 1.8.5, M43 gave me no output at all.  Didn't crash or hang.  Just nothing.

Time to investigate.

Bob-the-Kuhn, 2018-01-13:
I screwed up.  I was compiling the wrong directory when I said 1.8.5 had problems.  The Arduino 1.8.5 compiled version works as expected.

Anything else needed to reproduce the problem?

dembach, 2018-01-13:
Same here, no output......
but my Board is also stuck after executing M43....
PlatformIO ? can you explain what the difference is ?

Bob-the-Kuhn, 2018-01-13:
Once I stopped screwing up I had no problems with M43  Acted properly and executed commands afterwards.  Using both PlatformIO & Arduino 1.8.5 to compile.

Right now the only suggestion I have is to download a new copy of bugfix-2.0.x.

dembach, 2018-01-14:
Hey @Bob-the-Kuhn  : 

just downloaded the newest .... and.... what should i say.... working ;-) M43 with output, no crash...

just perfect as it should be... i will close this here....

closed 2018-01-14 by dembach

Bob-the-Kuhn, 2018-01-14:
FANTASTIC!