MarlinFirmware/Marlin #9158: Probe y offset seems to have no affect

prahjister, 2018-01-12:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: --- Delta Autocalibration offsets seem to have no affect. X direction is perfect. Y is slanted
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1627648/Configuration.zip)

- Expected behaviour: ---Autocalibration results to vary
- Actual behaviour: --- Almost Identical results
- Steps to reproduce:
  - Do this Changing Y_PROBE_OFFSET_FROM_EXTRUDE from +26 to +13 has little affect
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

LVD-AC, 2018-01-23:
Delta calibration probes with the nozzle (carriages, effector) at certain positions, not the probe.

prahjister, 2018-01-23:
I was able to make it work with this jig I created. What is the intended purpose of an offset probe? 
[image]

LVD-AC, 2018-01-23:
To probe **bed positions** in G29 etc. <> to probe **effector positions** in G33

closed 2018-02-14 by thinkyhead