MarlinFirmware/Marlin #9159: Nozzle crashes on bed with BLTOUCH during G29

FabLab-du-Don, 2018-01-12:
https://github.com/MarlinFirmware/Marlin/blob/0640bd028215a01ff9a01a46d4dcb859ba208a20/Marlin/Marlin_main.cpp#L1694-L1710
In **setup_for_endstop_or_probe_move()** there is no instruction for _Enable the endstops (for endstop moves)_.
`endstops.enable(true) ;` is missing.
**gcode_G28()** works fine because endstops are enabled at the beginning. With gcode_G29(), BLTOUCH is not deployed, the nozzle reach the bed and doesn't stop.

I hope to be understandable

token47, 2018-01-14:
My BLTOUCH works ok most of the time (even with g29). But occasionally it does not get deployed and the nozzle crashes on the bed. Then, in about 1 to 2 seconds, it gets deployed and instantly triggers, making the head back again. I get this problem in about 1 print every 10 prints or so. And it happens mostly only one time in the 16 probes (4 x 4 grid).

FabLab-du-Don, 2018-01-14:
I suspect the hardware to be guilty. The first spot of the G29 is very close to the edge. It pulls on the wire and the BLTouch lost the contact. 
I will solder my connectors and try again.

FabLab-du-Don, 2018-01-30:
After soldering, ther is no more problem.
Sorry

closed 2018-01-30 by FabLab-du-Don