MarlinFirmware/Marlin #9162: [FR] Redundant bed temperature sensors

echo465, 2018-01-13:
Currently temperature sensor types are defined in configuration.h like so:
```
#define TEMP_SENSOR_0 1
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_4 0
#define TEMP_SENSOR_BED 1
```

TEMP_SENSOR_1 can be defined as redundant to TEMP_SENSOR_0 for safety:
```
#define TEMP_SENSOR_1_AS_REDUNDANT
#define MAX_REDUNDANT_TEMP_SENSOR_DIFF 10
```

Unfortunately, there doesn't seem to be a way to define a temperature sensor to be redundant to the bed temperature sensor or any other sensor, it can only be configured that TEMP_SENSOR_1 is redundant to TEMP_SENSOR_0.  In the interest of increased safety, there should be some way to define a sensor to be redundant to TEMP_SENSOR_BED. More broadly, it would be 'nice to have' for one sensor to be redundant for any other sensor.

My motivation - one of my printers had TEMP_SENSOR_BED vibrate loose and land in the insulation underneath the bed while printing with a bed temp at 60c.  The printer responded by increasing power to the heater and the heater was able to keep the sensor reading 60c, so "Thermal Runaway" wasn't triggered.  Eventually, I smelled burning and cut power.  I measured bed temperature at 125c with an infrared thermometer several minutes after cutting power.

Spawn32, 2018-01-13:
To bad there isn't a up-vote button her :)

Having a redundant sensor to the bed would be nice, evne better if we could you use all temp sensors as redundant :)

panecki, 2020-05-19:
This should be one of the basic functions. Safety first 👍

slowbro, 2021-06-04:
I'm trying to give this a go, as it's something I was wanting recently, as well.

thisiskeithb, 2021-06-11:
#22085 has been merged, closing.

Thanks for implementing this feature @slowbro!

closed 2021-06-11 by thisiskeithb