MarlinFirmware/Marlin #9166: [FR] LCD selected menu color inverted (as before)

spilz87, 2018-01-13:
Hello

As I got issue with fullLCD to select a menu : 
When I go to fast with encoder, there is only one line ( instead of 2) so I can't know if the selected menu or life is up or down the line :(

Is there a way to have as before the selected menu in invert color (instead of the 2 lines) ?

Regards

Sebastianv650, 2018-01-13:
I also experienced this missing line or even more than 2 visible lines in recent Marlin versions. But I think we should fix them instead of going back to the inverted version. The reason for this change was performance, the lines are much faster than the inverted text.

I'm quite sure we had them working around 1.0 and 1.1, maybe something got changed since then?

spilz87, 2018-01-13:
I don't remember when was the change 
Is it as much faster ?
Because inverted was most "beautiful" (my opinion) 

Actually I find LCD menu more and more slower (compare to old version) and got issue when I navigate in sd card's files name : bad refresh, files missed, need to go down then up to see them, etc

AnHardt, 2018-01-13:
````
  // Enable to save many cycles by drawing a hollow frame on Menu Screens
  #define MENU_HOLLOW_FRAME
```

spilz87, 2018-01-13:
It works, but it's true it's much slower :(

closed 2018-01-13 by spilz87

AnHardt, 2018-01-13:
I don't remember the numbers for the menu, but at the info-screen the difference was ~3ms. Compared to the time the scren neds ro draw (below 30ms) this was 10%, just for reversing the bar.