MarlinFirmware/Marlin #9167: Problem with Z Offset

Hamztaly, 2018-01-13:
Hello everyone,

I have a problem configuring my ne bl touch sensor on my anycubic prusa i3. I downloaded the firmware from the anycubic website and modified it as described in tutroials.

Now the problem is that the nozze cannot be set with an offset value - you can see it here:
https://www.youtube.com/watch?v=1zSpnr5L7YE

so i cannot get the nozzle closer to the heatbed..

Do you have any suggestions how to solve this problem?

Thanks in advance

[AnycubicForI3.zip](https://github.com/MarlinFirmware/Marlin/files/1628372/AnycubicForI3.zip)

Roxy-3D, 2018-01-13:
I'm a little bit skeptical of your line:

  #define Z_PROBE_OFFSET_FROM_EXTRUDER -4 //-19.

I try to get my BL-Touch probes as high as possible so I have as much clearance as I can get while printing.   Even so...   Because of where the trigger point is they are only about 1.5 mm above the nozzle while printing.     My BL-Touch Z_PROBE_OFFSET_FROM_EXTRUDER number is under 1.mm

You are using a very old version of Marlin.   For the last year there is a configuration.h setting to declare (and setup) BL-Touch probes.    And there are all kinds of sanity checks in the Marlin firmware to make sure the BL-Touch probe is doing the right thing at various points in time.   You don't have any of those features because your code is too old.

You really need to move to the current Bugfix-v1.1.x code base.   That will help us give you the support you want.

Hamztaly, 2018-01-13:
Thank you for your answer. I would really like to upgrade to a newer version But i dont have the experience and know how to configure the whole firmware from the scratch.
Unfortunately anycubic does Not provide newer versions

Roxy-3D, 2018-01-13:
Install a visual diff program.    Notepad++ with the visual diff plug in turned on will do nicely.
Pull your old configuration.h  (and later configuration_adv.h) into it and compare against the default configuration.h file for the latest Marlin bugfix_v1.1.x.     Just work your way through the file making sure your settings are crossed over to the new file.  

You will find additions and name changes.   Cross your old values over to the changed names.   And just ignore the additions.    That probably will get your printer up and running with the new code.   And if not...   You will be very close and can post follow up questions here.

Hamztaly, 2018-01-13:
Okay thank you for the reply. This was also the way I thought I can do it. I will try it soon. For sure there will be other problems...

closed 2018-02-11 by thinkyhead