MarlinFirmware/Marlin #9169: [QUESTION] HEAT BED

calculateWorld, 2018-01-13:
When I heat the bed, boards feed back like that.
Error:MAXTEMP triggered, system stopped! Heater_ID: bed
[ERROR] Error:MAXTEMP triggered, system stopped! Heater_ID: bed

Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

The MAXTEMP was set higher than the current temperature.

revilor, 2018-01-14:
So what is your setting for MAXTEMP, and what is the bed target temperature when MAXTEMP triggers?

calculateWorld, 2018-01-14:
lt sent me this ERROR when i just started heat. The BED_MAXTEMP was set 150.  Bed temperature sensor was running normal. Different with the bed，the printer nozzle works.

revilor, 2018-01-14:
So you get the MAXTEMP error when heating the bed starts? That's weird. Please attach your Configuration.h and Configuration_adv.h files.

AnHardt, 2018-01-14:
Likely a short in the thermistor wires.

calculateWorld, 2018-01-15:
Yes, and to test this error, my configuration basically didn't being changed.   
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1630553/configuration.zip)

calculateWorld, 2018-01-15:
The thermistor works well, i can get the right bed temperature from it.

calculateWorld, 2018-01-16:
@lrpirlet
 The delta changed this setting , which I didn't noticed. But I try to set it to 10 and then to 100, it still doesn't work. If this is the reason, why it respond me the MAXTEMP error. I thought it should sent me Heating failed. Because if i turn off the power, i can get the Heating failed error after about 10 seconds when i start heating the printer nozzle.

revilor, 2018-01-16:
AnHard already suspected a hardware issue and I'm also quite sure about that. 

So what kind of heated bed are you using? And just to see what happens, can you please disable THERMAL_PROTECTION_BED and post the temperature readings you get when heating the bed. As the thermistor readings are not reliable don't turn the bed on for too long. 30 seconds to one minute should be enough.

calculateWorld, 2018-01-17:
@revilor 
Despite that i put "//" forward the THERMAL_PROTECTION_BED,  i still get the same error.  And this makes me more confuse.
Error:MAXTEMP triggered, system stopped! Heater_ID: bed
[ERROR] Error:MAXTEMP triggered, system stopped! Heater_ID: bed

This is the message from printer before i start heating the bed, in order to test the thermal.
SENDING:M105
ok T:18.59 /0.00 B:18.71 /0.00 @:0 B@:0

After i start heating the bed with my  **HAND**, the temperature rise.
SENDING:M105
ok T:18.48 /0.00 B:20.27 /0.00 @:0 B@:0

Otherwise i try to set the bed temperature from 0 to 30,  and under 17 is safe, up 20 will cause the error.

ps: I'm not quite sure what you say about the kind of bed.  My bed is a aluminium plate with resistance wire.

revilor, 2018-01-17:
Ok, my mistake: THERMAL_PROTECTION_BED can't be disabled because you  are not using PID control for the bed which means THERMAL_PROTECTION_BED has to be on in any case.

Actually I'm out of ideas. Besides the MAXTEMP trigger, everything looks reasonable. Bed measures room temperature, when heated the bed temperature rises. So far so normal. 

My last tip would be to either check thermistorTable_1.h if the file somehow got corrupted, or even better check out a clean Marlin bugfix-1.1.x with all the latest fixes, put your configuration in and see what happens. Hopefully it will work afterwards. Good luck!

closed 2018-02-11 by thinkyhead

GiyHub127754, 2019-03-17:
I had the same issue. Turned out that where the heat bed plugs into the controller box the connector used was under rated and over many uses the connector heated up/cooled down and eventually formed a dodgy joint...

sporoisty, 2019-03-17:
After a long time, I started to test my board, and then I found it got an error in heating valves. So I replaced the board with a new one, with that operation, my issue had gone finally . I have to say found this problem troubled me a lot. I'm not sure my experience will help you or not. @GiyHub127754