MarlinFirmware/Marlin #9171: [1.1.8] Experiencing Issues with: LCD Bed Leveling - probe manually - 3 point level

RossKor, 2018-01-13:
Sometimes the process hangs on "Homing XYZ"

Sometimes (once) the LCD blanked after finishing homing, i may have pressed the button before it finished homing, the screen refreshed when i turned the knob.

Sometimes (most, if not all, of the time actually) after accepting the 3rd point i get a MAX_TEMP error if i'm leveling while hot.

Sometimes after completing the bed level process the printer hangs when i try to home it.

After one successful leveling process I attempted to start a print and the printer just crashed into all the axis (hence my attempts to run auto-home after leveling).

I have NEVER gotten to the point of starting a print after completing the leveling process. I've been printing fine prior to trying to enable this functionality but i'm sick of constantly leveling the corners as on my printer removing parts almost always knocks the bed a bit out of whack :/

Using
#define LCD_BED_LEVELING
#define PROBE_MANUALLY
#define AUTO_BED_LEVELING_3POINT

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/1628676/configs.zip)

_I've had to make additional "configuration" changes outside the config and config_adv files which are not included in the zip._

thinkyhead, 2018-02-11:
Sometimes it "hangs" waiting for an extra click.

Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

boelle, 2019-02-20:
@RossKor Please test with the latest  `bugfix-2.0.x`

boelle, 2019-03-06:
@thinkyhead i think we can close this one

boelle, 2019-07-21:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle