MarlinFirmware/Marlin #9172: [FR] Dual Z Stepper Dual Extrusion Auto Level Slope Changes

skc5741, 2018-01-13:
Hello,

I've been running into an issue with auto-leveling, relating to my Chimera Dual-extrusion hotend. It has two nozzles, and neither one is retractable. It looks like this:

[image]

When I try to make a part, the secondary nozzle occasionally will occasionally pull up a sting of plastic that the primary nozzle just laid. My Z system looks like this:

[image]

The Z is controlled by two steppers. It also has a small amount of slack, so that I can turn one lead screw so that one side of the Z raises, and the angle of the nozzle is changed. When I run into this problem with the secondary nozzle, I turn the lead screw closest to the secondary nozzle up so that the secondary nozzle is raised relative to the primary nozzle.

When I want to use the secondary nozzle for support material, I'll have to adjust one of the lead screw so that the two nozzles will be exactly the same height off of the bed. But my bed is not perfectly flat, and at certain positions on the bed the slope is different than others. If the auto leveling feature has already found the slope of the bed at any given point, why can't it adjust one of the lead screws so the nozzles can match that slope, and each nozzle will be exactly the same distance above the bed?

What do you guys think? Is there any way I can make this happen?
Sam

skc5741, 2018-01-14:
Hello,

Quick update, I'm trying to implement a feature that would level the X carriage before beginning a print, using an unused gcode command G34. I got as far as the program reporting how far the Z offset is between one side of the bed and another, but I need to make the final adjustment with the carriage and I'm running into a wall.

If I want to move the Z axis, I set `destination[Z_AXIS] = ` to whatever I want to, but I just want to move the second stepper, Z2, the amount that the offset reported. What is a command I can use in Marlin_main.cpp to move the second stepper, Z2, a specific distance?

Thanks,
Sam

thinkyhead, 2018-02-11:
When using that kind of dual nozzle, your setup needs to be very flat and rigid. It is recommended to not use a bed that moves in Y, but only a bed that moves in Z. No, we are not likely to ever implement compensation by independently tweaking the Z leadscrews.

closed 2018-02-11 by thinkyhead