MarlinFirmware/Marlin #9175: [2.0.x] CR 10 stock, having display problems

Dorwido, 2018-01-14:
Using the cr 10 stock board and display, working fine with 1.1.x-patch branch, currently trying to get it running with 2.0.x, kinda using the example config and also tried quite some stuff to get it running, but whatever I try the display is just off.

Maybe someone with more experience can point me to the right direction.

[image]

AnHardt, 2018-01-14:
```
// If you want you can define your own set of delays in Configuration.h
//#define ST7920_DELAY_1 DELAY_0_NOP
//#define ST7920_DELAY_2 DELAY_0_NOP
//#define ST7920_DELAY_3 DELAY_0_NOP
```

Dorwido, 2018-01-14:
Tried thing there now too without success, just to be sure it looks like that all time, even compared again to the 1.1.x-patch branch and my configs etc...keeps looking like that in the video

[https://youtu.be/pE8slGLoAJM](url)

thinkyhead, 2018-02-11:
Theoretically, Marlin 2.0 (AVR) should behave identically to 1.1.

Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

Dorwido, 2018-02-13:
Currently no time to test it again, but when I made the ticket the bugfix-1.1.x was working fine, it was just the 2.0 branch I couldnt get the lcd display to work proper.

ldkraemer, 2018-02-15:
I have Bugfix Version 2.0.x running on my Creality CR-10 Mini.  I had to change the following to
clear up the same problem with my Display.

~/path/to/Marlin-bugfix-2.0.x/Marlin/src/lcd/dogm/ultralcd_st7920_u8glib_rrd_AVR.cpp

// set optimization zo ARXDUINO optimizes this files
//#pragma GCC optimize (3)

Make sure you also have the following selected in the configuration.h file:
#define MOTHERBOARD BOARD_MELZI_CREALITY
#define CR10_STOCKDISPLAY

It should built and everything should work, EXCEPT for accessing the SDCARD.

easyconfig version 1.1.x is the last version that has a working SDCARD, as far as I can determine....

These are the Components in my CS-10 Mini
https://www.ebay.ie/itm/Creality-3D-CR-10-3D-Printer-Control-Mainboard-Upgrade-Circuit-Board-Control-/253193655402
https://www.aliexpress.com/item/Universal-LCD-12864-3D-Printer-Display-Screen-With-Encoder-For-CR-10-CR-7-Model/32833148327.html

[_Statusscreen.txt](https://github.com/MarlinFirmware/Marlin/files/1728167/_Statusscreen.txt)

I'm using this Statusscreen for my Mini.

If you are trying to use a bootscreen you will need to also make these changes:

~path/to//Marlin-bugfix-2.0.x/Marlin/src/lcd/dogm/dogm_bitmaps.h
Line 40
    #include "../../../_Bootscreen.h"
Line 136
  #include "../../../_Statusscreen.h"

Larry

AnHardt, 2018-02-15:
The ability to use the ST7920_delays was restored yesterday. #9624

https://github.com/MarlinFirmware/Marlin/issues/9255#issuecomment-365598119

thinkyhead, 2018-02-23:
Sounds like this is solved. If not, we are happy to re-open it.

closed 2018-02-23 by thinkyhead

leytanelectronics2, 2019-01-17:
> ```
> // If you want you can define your own set of delays in Configuration.h
> //#define ST7920_DELAY_1 DELAY_0_NOP
> //#define ST7920_DELAY_2 DELAY_0_NOP
> //#define ST7920_DELAY_3 DELAY_0_NOP
> ```

im trying to find those but i am having no luck, im even using the find option from the edit tab and putting the names in but im not getting anything, im using marlin 1.8.8.

guecko333, 2020-03-22:
Hi everybody, sorry for my english, i'm french.
I had the same issue when i had just install a new firmware on my ender 3 SKR mini E3 1.2. There is some strange characters and display.
I resolve this problem by clicking on Restore Failsafe on my Ender 3 configuration and issue disappear. (it clear some old datas ...)
Hope it will be helpfull