MarlinFirmware/Marlin #9176: Motors getting too hot

batata004, 2018-01-14:
Hi,

I am using the latest marlin firmware (1.1.8) and for some reason my X and Y motors are getting veery hot, I can only touch them for 2 seconds at maximum. My prints are coming out very good but I dont think that this temperature is ok.

I already calibrated my stepper current exactly as it should be. So my question is: do you think chaning DEFAULT_MAX_FEEDRATE or DEFAULT_MAX_ACCELERATION in Configuration.h should solve the problem, I mean, could it reduce the temperature of the motors?

fiveangle, 2018-01-14:
When activated, the temperature of the stepper motors is almost directly related to the configured  stepper driver current since all current going into the motor, even when "idle" simply heats up the windings, or does useful work.  Only if your stepper drivers are controlled by Marlin via DAC or PWM could the firmware affect this.  You said you calibrated your stepper current, so I assume this was using pots on the drivers ?  If so, Marlin isn't part of the equation.

batata004, 2018-01-14:
Yeap, I use the trimpots on each A4988 driver to get the minimum current. So I will keep investiganting and close this thread since you said marlin is not the curlprit! Thanks!

closed 2018-01-14 by batata004

AnHardt, 2018-01-14:
a. You could disable the steppers. But for X and Y this will have a low cooling effect.
b. You could lower the allowed acceleration (and or jerk) settings. This theoretically allows lower currents - less heat.
c. You could try to lower the friction an the axis to reduce the needed current. (Better alignment, better barings, more oil or grease, ...)