MarlinFirmware/Marlin #9177: Int Sd card problems?

Comuter83, 2018-01-14:
I need help here with one thing. I convert a Cr-10s 500 to marlin frimware and E3D Vulcano. I have problems with the SD card reader. In the menu there is "no SD card". though it's a card in. If I press "no SD card", all files on the card will be displayed but the printer will not start warming when I select a file. but when i enter the serial monitor and send the command "M21" (INT SD CARD). then I can access the SD card and choose file as usual. and then start the printer. So what can this be?

GMagician, 2018-01-15:
sd card insert/remove detection is done by an I/O signal. if detection doesn't work it can be:
1) your lcd/board doesn't handle such I/O
2) I/O is damaged
3) your card reader detection switch is broken (it's part of the metallic slot)

M21 is present to start SD initialization when such "auto detection" doesn't work (recently has been re-added also a menu item to execute M21, if I'm not wrong).
Maybe there is some block in firmware that limits SD usage until it's properly initialized, that's why you have this issue.

Comuter83, 2018-01-15:
Ok! thanks for your help. Some ppl are saying that it is the 1.1.8 firmware that is causing this problem. so i think i will try with 1.1.6 version. To see if its working properly there.

Comuter83, 2018-01-15:
I think i solved the problem in one way. And one way not solved it. It seems to be newer hardware parts in my printer than the CR-10 S5 there are som differences on the extruder and filamentsensor as well. the extruder are now black instead of yellow and have a creality 3D logo on it. And the fillasensor is now mold injected plastic instead of 3D- printed. so im not the man to solve this problem. :(

programmercert, 2018-01-17:
I did have the-same issue few weeks ago when I updated Marlin on my CR-10 S5 but I fixed it.

HERE IS THE FIX:

1.Comment out or disable "#define SD_DETECT_INVERTED" in Configuration_adv.h
2.Find the first instance of "#define SD_DETECT_PIN" in pins_RAMPS.h and change the value from 49 to -1.

GMagician, 2018-01-17:
@programmercert & @Comuter83  your solution confirm my point 1. System doesn't have I/O to handle sd detection. This means that M21 is mandatory to use SD access (or by menu item)

programmercert, 2018-01-17:
@GMagician  The pin stuff you talked about may be right but "M21" is not required. I just refresh it in the menu that's that.

GMagician, 2018-01-18:
@programmercert that's not completly true. M21 calls an internal function that after initialization speed up bus and also does some initialization for newer SD card types (and activate also CRC).
Maybe it can works without but it's not the best

Comuter83, 2018-01-18:
okay! tnx for all answers! @programmercert i will try this when ive got time and return and say how it did go for me. @GMagician im completely satisfyd if i just can run the M21 from the printer, like a menu btn "INT SD CARD"

GMagician, 2018-01-18:
@Comuter83 don't remember PR number but item has been added. Once you will set 'SD_DETECT_PIN' to -1 a 'Init. SD card' will appears

thinkyhead, 2018-02-11:
Bummer. My CR-10S 300 has no problem seeing the SD Card insertion and removal.

closed 2018-02-11 by thinkyhead

programmercert, 2018-03-10:
@thinkyhead  It's wrong to close this. OP is using _Cr-10s 500_  not _Cr-10s 300_ . I had the-same issue with OP too and many on other website also have it. Only few people have Cr-10s 500 and think of modifying the firmware. Those that will modify the firmware will run into this issue since it's not fixed.

thinkyhead, 2018-03-11:
@programmercert
Please file a new issue, filling out the requested details, and someone will look into it.

programmercert, 2018-03-11:
@thinkyhead  That's fine. I have already fixed it and replied to the OP with instructions on how to fix it and I also expected it to be fixed in the source with my comment or which other way you all find to be a better fix. There is no need to create another of this question again. It will be fine if you re-open this issue instead and then close it when it is fixed.

thinkyhead, 2018-03-11:
Sounds like what we need is the contribution of a proper CR-10S 500 configuration. If you have well-tuned ones, please post them in a ZIP and we'll be able to include them in 1.1.9.

LeererMann, 2019-11-17:
> I did have the-same issue few weeks ago when I updated Marlin on my CR-10 S5 but I fixed it.
> 
> HERE IS THE FIX:
> 
> 1.Comment out or disable "#define SD_DETECT_INVERTED" in Configuration_adv.h
> 2.Find the first instance of "#define SD_DETECT_PIN" in pins_RAMPS.h and change the value from 49 to -1.

This fixed the problem for me in Marlin 1.1.9 on a CR-10 S5.

In the menu it said "No SD Card" but I could click on it and see the files on the SD card. I could even pick a file and the elapsed time clock started counting up but the bed & hotend never heated up so it never started printing.