MarlinFirmware/Marlin #9182: Several notes from the last several days troubleshooting mks sbase issues.

Zakm0n, 2018-01-14:
So, my sbase board is currently functioning perfectly with a build I downloaded on 1/8. There is something that has been added to the code since then that has screwed up my serial connection. I've noticed that on the newer builds that have the serial crash, whenever I send M503, I get an incomplete return. The bottom half of the output gets dropped for whatever reason. With the older version, the entire output gets shown. I had issues at first with the older version, because I couldn't write to eeprom, but using a different sd card fixed that for whatever reason. 

TL;DR: something added to the 2.0.x bugfix code in the last 5 days has borked serial for my mks sbase, and sbase eeprom can be picky about the ad card used to hold the FW.

p3p, 2018-01-15:
Any chance you test the commit before #8148 was merged (https://github.com/MarlinFirmware/Marlin/tree/cf654b9a1aadb9dcae6c6c7f5dff4c74618494b5)
and the one after (https://github.com/MarlinFirmware/Marlin/tree/d7fd78cb9136a3f42244e965441ac3c5625a10ac)

As for the sdcard they can be a bit flaky, but if it works with Smoothie it should work with Marlin..

(related #9170)

Spawn32, 2018-01-15:
I have used https://github.com/MarlinFirmware/Marlin/pull/8148 since early november without any problems on my MKS Sbase, so would be surprised, but you never know :/

closed 2018-02-11 by thinkyhead