MarlinFirmware/Marlin #9184: RAMPs restart when Z home  whit  DEFAULT_AXIS_STEPS_PER_UNIT 12800

jcrorxp, 2018-01-15:
------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

my print board is ramps 1.4  mega 2560 .   EEPROM undefined.

when i use the Printrun to home the z axis,  ramps will restart again and again  whit this configuration : 

#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 6400, 500 }
#define HOMING_FEEDRATE_Z  (1000)
or 
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 12800, 500 }
#define HOMING_FEEDRATE_Z  (300)

it will not restart whit this configuration:
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 6400, 500 }
#define HOMING_FEEDRATE_Z  (200)
or
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 12800, 500 }
#define HOMING_FEEDRATE_Z  (100)

is the problem on AVR PWM  frequency ? may be unsupport high frequency?

my driver is LV8729 ,  max 128 STEPS.

GMagician, 2018-01-15:
I'm not sure but I think mega2560 has limits to handle more than 32 microsteps. It's not so powerfull to be able to generate step/dir signal at such high freq (doing all other work in the same time)

fiveangle, 2018-01-15:
Your main viable options are to decrease microstepping on your Z (probably to 1:1), reduce max feedrate for Z axis to where it doesn't saturate the MCU, or switch to  32-bit board.  Changing your Z mechanicals to allow lower steps-per-unit is probably best.

jcrorxp, 2018-01-16:
we ues it to make a LCD UV 3D printer  just Z axis used。
we try to use  32 microstep in  a delta 3D printer , mega2560 , not very well~ 16 microsteps is better

closed 2018-02-11 by thinkyhead