MarlinFirmware/Marlin #9187: Z offset not same on all bed using Bilinear

VincentWiz, 2018-01-15:
### Bug Report
- Description: On Marlin 1.1.8, using capacitive sensor and Bilinear autolevel method, the bed isn't flat everywhere. A discussion has been open in a group on Facebook, and many people are having this issue. I tried to clear EEPROM then re-upload Marlin, no effects. I also tried to go on Marlin1.1.6, same. Other people has the BLTouch, so the issue isn't comming from the sensor. Some people are also on Marlin 1.1.6. I wanted to use the UBL for checking, but i has a compilation error (Text overflowed). I'll post my configuration.h and conf_adv.h this evening.

- Expected behaviour: Autolevel should compense the Z offset everywhere on the bed

- Actual behaviour: Autolevel doesn't compense the Z offset everywhere (check the picture attached)

[image]

- Steps to reproduce:
  - Unable to define

vincentpfarr, 2018-01-15:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1632584/Marlin.zip)

Here are the files :)

vincentpfarr, 2018-01-15:
My bed is almost flat, here is the G29 T : 
SENDING:G29 T
Bilinear Leveling Grid:
      0      1      2
 0 +0.024 -0.089 -0.234
 1 +0.091 -0.093 -0.385
 2 +0.172 -0.105 -0.521

This is VERY weird to get those values. Left side is going up, but right side is going down.

vincentpfarr, 2018-01-15:
The G29 T from another guy : 

Bilinear Leveling Grid:
      0      1      2
 0 +0.614 +0.103 -0.456
 1 +0.467 +0.041 -0.383
 2 +0.367 -0.101 -0.461

benlye, 2018-01-15:
Are you sure that there is nothing affecting the geometry of your printer as the probe goes to the back corner?  Cable tension pulling the print head up?  X or Y rods not straight?  Z lead screws out of sync? 

What kind of printer is it?

VincentWiz, 2018-01-16:
Hello, 
It's a Anet 18 (Prusa I3).
I checked all cables, all rods are straight (was my first idea). 
Z are at the same level (measured with a sliding caliper. 
My bed is new, and i tested with a Anycubic glass, same issue.

benlye, 2018-01-16:
Cool, I have an Anet A8 as well.  Bi-linear leveling works well for me.

You should probably try probing more points - I'd try at least 5x5.  On my printer I use 7x7.

VincentWiz, 2018-01-16:
I also tried the 5x5, had the same issue (and is very long when you have 3 check per point xD)

I can try the 7x7 this evening. Do you have triple check ? Is there a way to not G29 at each print?

Thanks

benlye, 2018-01-16:
There are a couple of things you can do to speed it up:
* Disable `#define MULTIPLE_PROBING 2` (I found it makes no real difference)
* Change `#define Z_CLEARANCE_BETWEEN_PROBES 5` to `#define Z_CLEARANCE_BETWEEN_PROBES 2` 
* Change  `#define XY_PROBE_SPEED 6000` to `#define XY_PROBE_SPEED 8000`

I would also check these two values:
```
#define X_PROBE_OFFSET_FROM_EXTRUDER -26
#define Y_PROBE_OFFSET_FROM_EXTRUDER -40
```
Did you measure those yourself, or were you given them by someone else?  It is extremely important that they are accurate.  Being out by 1-2mm is known to cause the phenomenon you're seeing.

And finally, yes, there is a way to store the G29 grid so you don't have to do it at the start of each print - run `G29` once then `M500` to save the settings to the EEPROM.  Then at the start of each print instead of G29, run `M420 S1` to activate the saved grid.  I personally don't do this, just in case I've upset my bed when removing a print or replacing the tape.  With the tweaks above my 7x7 probing doesn't take long at all.

VincentWiz, 2018-01-16:
I will test changes you proposed this evening. 

I didn't measure those values, these were given by the one who designed the probe support (Thingiverse). How does this affect the Z-Offset as they are X/Y offset?

Thanks for the bed levelling. You are right, each time i try to remove the print, the bed would move. 

Thanks very much

benlye, 2018-01-16:
> I didn't measure those values, these were given by the one who designed the probe support (Thingiverse). How does this affect the Z-Offset as they are X/Y offset?

When the printer measures the bed it needs to know where it took that measurement so that when the nozzle is at that point it knows how much correction to apply.  If the X/Y offsets are incorrect then the correction is applied in the wrong place.  I had the exact same problem you have, and it was caused by my Y offset being out by 2mm, because I was using the measurements given by the support designer on Thingiverse (looking at your offsets, I suspect we might be using the same one).

For an Anet with a 220mm x 220mm bed you can use these instructions to find your offsets:

1. Measure the bed and make a mark in the center (X110 Y110)
1. Home the printer then move the nozzle to X110 Y110 Z0
1. If the nozzle is not on the mark you will need to make small movements to get it exactly on the mark
1. With the nozzle on the mark in the center of the bed, measure the X and Y distance from the edge of the bed to the side of the probe
1. Measure the diameter of the probe
1. The probe offset values are `110 - [the distance from the bed edge to the probe] + ([probe diameter / 2])`

The offset parameters are integers, so round your results to whole numbers before updating configuration.h

For example, if you measure 85mm from the edge of the bed to the edge of the probe, and your probe diameter is 18mm, the equation is `110 - 85 + (18/2) = 34`

VincentWiz, 2018-01-16:
Ok I understand. Thanks for explanations

I will check all these parameters this evening, I will recalibrate will send results here.

will5517, 2018-01-16:
Hello, I'm sorry for my bad English but I use a translator.

once i have calculated with your method the parameters of decalage X and Y, how to insert them in marlin in the file configuration.H at which place ???
thank you in advance

benlye, 2018-01-16:
In Configuration.h you need to modify these two lines:
```
#define X_PROBE_OFFSET_FROM_EXTRUDER -25  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -41  // Y offset: -front +behind [the nozzle]
```

Pay attention to the comments, which tell you whether the number should be positive or negative.

vincentpfarr, 2018-01-16:
Hey, 

After : 
Disable #define MULTIPLE_PROBING 2 
Change #define Z_CLEARANCE_BETWEEN_PROBES 5 to #define Z_CLEARANCE_BETWEEN_PROBES 2
Change #define XY_PROBE_SPEED 6000 to #define XY_PROBE_SPEED 8000
Change probe points from 3 to 7

Exactly the same, right side could not be printed because the noozle was too close to the bed
[image]

I'll check the center

will5517, 2018-01-16:
OK BENLYE 

it is the X and Y min_post that I have to change
thank you very much

will5517, 2018-01-16:
for me too after all these changes (same as Vincent), my problem remains the same.

benlye, 2018-01-16:
OK, next question. How long do you preheat the nozzle for before printing? 

In my experience it takes a good 5-6 minutes for everything to finish expanding. If you start the print as soon as your nozzle is at temperature, it will be nearer to the bed for the later parts of the print.

will5517, 2018-01-16:
For my part the impression starts as soon as the nozzle reaches 210 ° and the bed has 60 °

benlye, 2018-01-16:
> For my part the impression starts as soon as the nozzle reaches 210 ° and the bed has 60 °

Then you should definitely wait a little while.  Try waiting 5 minutes from the time the nozzle is at 210 before you start printing.

will5517, 2018-01-16:
ok i'm testing this right away

benlye, 2018-01-16:
And just to prove that this really does work, this is a single 0.15mm layer  test pattern I just printed.
[image]

will5517, 2018-01-16:
here is my print lol not very successful no way to solve this problem
[image]
[image]
[image]
[image]
[image]

Roxy-3D, 2018-01-16:
> here is my print lol not very successful no way to solve this problem.

You solve this by getting the mesh point values defined better.   For what ever reason, the auto probed values are not good enough.   Try using M421's to shift those values so it can print.

will5517, 2018-01-16:
ok Roxy-3D   
 but i do not know how to do it ... can you explain me please

Roxy-3D, 2018-01-16:
It would be easier if you had UBL enabled.  It has an interactive mesh editor built in.   And the auto probing works pretty well in UBL so most likely this issue would not come up for you.

It looks like the nozzle is too close to the bed in the lower left corner.   Assuming that is the origin, you would issue M421 I0 J0 Z1.0 to raise that mesh point to a height of 1.0 mm.   

I think it would be wise to get a G26 Mesh Validation pattern printed.   It will be much clearer from that what needs to be done.

will5517, 2018-01-16:
ok but I can not activate the UBL on my card anet v1 more space than I could remove in the files configuration.H and adv .H so that it can fit in the eeprom ??? for the rest and adjustment of the ubl there is no tutorial ??? because I am a beginner and do not understand all that you explain to me.
thanks in advance

Roxy-3D, 2018-01-17:
There is good documentation at:  https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_features/unified_bed_leveling.md

But if you have a 128 KB processor...   You probably can't run it.   You can still turn on the G26 and print a mesh validation pattern.   That will be very helpful in editing your mesh points.

will5517, 2018-01-17:
Thank you Roxy-3D, but you're right it is not possible to activate the UBL with the anet card. By cons I managed to activate the G26 and the result is in the photo.
Is there no way through another way to solve my problem ???
thank you in advance

[image]
[image]

VincentWiz, 2018-01-17:
This is weird that both of people who have this issue has the right part of the bed too close of the noozle, and not the left for example.

will5517, 2018-01-17:
on my other AM8 is the opposite the left side is too tight

will5517, 2018-01-20:
good evening, after setting the ubl and have saved the map in the firm with the command G29 S1, what command gcode should we put in the script of the slicer, to load and activated the compensation with UBL ???
I tried the command G29 L1 or the command M420 L1 and removed the command G29, but after very short time, the printer goes up bottom Z stop I do not understand why!

Roxy-3D, 2018-01-20:
> I also wonder why the grid does not print on the front of the bed

It looks like the G26 command is printing a 'border' on all sides.   It kind of looks like you don't have the mesh dimension set to cover the whole bed.    And really...   You should be using a bigger mesh than 4 x 4.     Go 8 x 8 or 10 x 10.

Roxy-3D, 2018-01-20:
> good evening, after setting the ubl and have saved the map in the firm with the command G29 S1, what command gcode should we put in the script of the slicer, to load and activated the compensation with UBL ???

I choose to have UBL active when the power powers up or resets.  You do that by doing
- G28
- G29 L1  (Or what ever location you have your mesh stored at)
- G29 A   (To activate UBL)
- G29 S1
- M500

Until you change things...  UBL will be active at power up and you don't need any G29's in your Slicer's start up GCode.

> I tried the command G29 L1 or the command M420 L1 and removed the command G29, but after very short time, the printer goes up bottom Z stop I do not understand why!

Are you moving the nozzle off the mesh?   That is what it sounds like.

thinkyhead, 2018-01-20:
@VincentWiz If you have a chance, also test with `PROBE_MANUALLY` (in place of the `FIX_MOUNTED_PROBE`) and compare the mesh result. Does the mesh differ in any unusual way from the mesh produced by using the probe? Do you get a better or worse result from the G26 mesh pattern?

will5517, 2018-01-20:
I have a problem when I run a print, how to explain ... I put the video ...

http://ul.to/u7gc50s5

after I am obliged to make a stop print or the Z goes up to the stop !!

will5517, 2018-01-20:
Are you moving the nozzle off the mesh? That is what it sounds like.

no the calibration cube at the top right is in the mesh

will5517, 2018-01-20:
I do not know why the calibration grid does not print on the front of the bed
but it is already much better to print the calibration grid after the G29 P4 T controls and fine tuning
[image]

will5517, 2018-01-20:
yes you're right the print came out of the mesh

Now I wonder how to adjust the mesh so that it squares all the bed !! do you have an idea ??
thank you in advance

will5517, 2018-01-20:
when I stay in the mesh everything goes well !!!! 

I will try to solve the problem of the G26 that does not print on the front of the bed and then go to 8X8 point ...
[image]

Roxy-3D, 2018-01-20:
If you do a G29 T do you see dots along one or two sides?   If so, those are mesh points that never got measured.   You can fill those in with G29 P3 T      The firmware will put in 'reasonable' values that are good enough to do a G26 so you can edit those mesh points and get them correct.

will5517, 2018-01-20:
OK thank you very much here is the result

[image]
[image]

Perfect

but setting up for a novice like me is not easy

Roxy-3D, 2018-01-22:
Very good!    One problem you are fighting is the tape never goes down the same.    If you turn on the Z-Babystepping....  (DOUBLE_CLICK!!!)  You won't have a problem even with a big print that takes the whole bed.   You can just tweak the height while it is printing the first layer.   And once that is down, You are going to be golden.

will5517, 2018-01-24:
Hello Roxy-3D and HELP , I still have a problem with the establishment of the UBL.
I explain to you my procedure and the problem:

1 / I initialize the eeprom with this command: M502 M500 M501 then enter
2 / I'm doing a G28
3 / I do G29 P T
4 / I'm doing G29 P2 B T and I'm the procedure, or I'm also trying a G29 P3 T and the result is the same.
5 / I make a G29 S1
6 / I do G26 B60 H210 F1.75 L0.2 S0.4 the printing of the mesh starts and when the print arrives on the front corner of the printer and quite to the right, the printing stops because the Z go up to infinity I am obliged to kill the impression.

I have started ten times from my pc and even directly with the printer screen via the ulb step-by-step option.

do you have a solution to my problem because I am completely lost.

thank you in advance
[image]

thinkyhead, 2018-02-14:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

vincentpfarr, 2018-02-18:
@thinkyhead  is this bugfix for bilinear or UBL ? Because i had no fix for Bilinear, and the issue is still the same...

Roxy-3D, 2018-02-18:
> is this bugfix for bilinear or UBL ? Because i had no fix for Bilinear, and the issue is still the same...

I think we have determined that a large number of the problems being seen are caused by twisted X-Axis guide rails.   There is no software fix for that.    UBL was just suggested as a 'work around' until we figured out what was causing the problem.

The other thread has good information to understand the problem:  https://github.com/MarlinFirmware/Marlin/issues/9529#issuecomment-366497815

VincentWiz, 2018-02-19:
Hey,

Nice, Houseofbugs proposed something: 
"
Everyone!!! I have the fix!

One of my customers was messing with his machine and noticed the Y belt tension was too much. He loosened it a bit and it fixed the issue. I just tested this on my 2 machines having the issue and IT WORKED ON THE FIRST TRY.

Loosen your Y belt tension. The belt was pulling on the carriage during printing causing the bed to skew."

And ofc, i just installed the Y belt tensionner so.. Maybe this issue is comming from there. I'll check this evening.

filipgoc, 2018-02-20:
> twisted X-Axis guide rails. There is no software fix for that.
Haha :-D 
No kidding.

VincentWiz, 2018-02-20:
I don't understand why, even is the rail is twisted, the sensor would take the distance, and will be closer to the bed where the rail is twisted, but the software will compense the twist no?

Thanks

AnHardt, 2018-02-20:
@VincentWiz 
Our bed tramming algorithms try to keep the distance bed-probe constant. This works pretty well as long the distance between nozzle and probe is constant. X-rotation prevents that https://github.com/MarlinFirmware/Marlin/issues/9529#issuecomment-365298112 and follwups made this a bit clarer to some.

@filipgoc 
Right. Almost everything can be fixed in software. The question here is if it can be done in the time we have and where can we get the needed data. This is a nonlinear problem. Until someone finds a trick we'd have to calc a sin() and or cos(). Because a similar problem exists for Y the amount of calculations will grow fast. (An idea could be to apply the correction not to every point we go, but only to tune the mesh (if we have one)). An alternative for hand tuning the mesh. While thinking about it - hand tuning the UBL mesh is the procedure to get the required data.)

@thinkyhead 
Somewhere else you talked about a test to compare the probes result with a nozzle-paper-test in the mid of the bed to find the `Z_PROBE_OFFSET_FROM_EXTRUDER`. For debugging purpose a way to make this test at X Y or at the current position would be nice.

filipgoc, 2018-02-20:
@AnHardt Good point. Though, at some point, the real question does become what is desirable to fix in software, as you hint towards. A dramatic rephrasing of the thought: 

"We fixed it, sonny."
(Long dramatic pause. Camera zooms in on the programmer's face, visibly in pain. Trying not to cry.)

"But _at    what    price...._"

(Breaks down crying. Camera gently pans away and up into the sunny sky that offers a stark contrast to the emotional darkness of the scene. Cut.)

Roxy-3D, 2018-02-20:
> The question here is if it can be done in the time we have and where can we get the needed data. This is a nonlinear problem. Until someone finds a trick we'd have to calc a sin() and or cos(). [...]

If we decided to fix this in software...   We probably don't need to do the sin() and cos() computation at each point.   We could apply the mesh concept to the twist also.   At any given location, we could be within a mesh cell that holds accurate sin() and cos() values for the twist at that location.   And probably, we could do the interpolation.   (It maybe it doesn't even need to be bi-linear.   It maybe the bulk of the correction can be done on each individual axis.)

We would also have to figure out accurate ways to measure the twist so the correction data could be loaded and used...

For right now...   It seems more reasonable to have the requirement the machine is built properly.

thinkyhead, 2018-02-25:
When we add the "too much deviation" test, Marlin will simply reject a mesh that is obviously wrong. But I fear there is no general solution to work around a weird gantry or off-angle probe. Sometimes the little wheels that the carriage rides on are a little off, and then your mesh comes out "lumpy." I don't think we can fix that automatically either.

filipgoc, 2018-02-25:
The 'too much deviation' reminds me of missing points. Sometimes, points end up missing and can be easily filled in with G29 P3. Maybe a warning that there _are_ missing points and that G29 P3 (or redoing the probing) would be prudent. If one runs with some terminal and can see the topology, this is not that big a deal, but over the LCD, it's not easy to see if the mesh is all filled in. Or is there an easy way I'm missing ?

filipgoc, 2018-02-25:
A follow up question to that... *why would points be missing in mesh?* If I do automatic probing, the probe does correctly sense the bed, and it works most of the time, why would every now and then the probing come back with one or two missing points in the mesh? 

I have observed this behavior, but didn't realize it may be a bug or something.

thinkyhead, 2018-02-25:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-1.1.x` or `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

You can use `M114 D` as you move around the bed to see lots of details about the current position. If you see strange things, also copy them into your text file.

Since you're using `BILINEAR` leveling, you can also use `G29 Q` at any time to see details about leveling at the current position.

We look forward to your next report. Hopefully we can narrow this down to a single cause and fix it.

filipgoc, 2018-02-25:
Sorry, I got threads confused. The missing points fun is from UBL. (I'm using/ testing both UBL and Bilinear ABL).

closed 2018-03-07 by thinkyhead