MarlinFirmware/Marlin #9190: [RFC] Lightweight ReprapDiscount Full Graphics Smart Controller GUI

marcio-ao, 2018-01-15:
### Request for Comment:

Currently the Marlin support for the ReprapDiscount Full Graphics Smart Controller 128x64 display is based on the U8G library.

However, the ST7920 chip used in that display module has a built-in character generator and an interesting (undocumented) compositing feature that allows graphics and text to be intermixed. As a proof-of-concept, I made an [Arduino sketch](https://code.alephobjects.com/diffusion/CHIPLIB/browse/master/arduino-sketches/reprap-st7920-ui-mockup/reprap-st7920-ui-mockup.ino) that demonstrates a simple 3D printer user interface:

[image]

The only graphical element is the progress bar rectangle and shaded portion, the rest is done using the text buffer which is conveniently XORed together with the graphics buffer by the chip. The nozzle, bed and rotating fans also use the built-in character generator, so animating the fan requires a SPI transmission of only two bytes. This weekend I also put together a [teaser video](https://www.youtube.com/watch?v=x7Vzw9quQ2I&feature=youtu.be) for an Instructable I plan to write on the topic, which shows more of the capabilities.

Anyhow, I know Marlin already has a graphical UI that works, but I thought I would ask whether there would be any interest in pursuing this new one. Some of the advantages:

- It would free up the CPU and program memory currently used by U8G and the associated fonts.
- It would speed up the GUI and cut down on SPI traffic, as the majority of the status screen updates would be done using ASCII characters rather than bitmaps.
- The larger font is a lot more legible, IMHO.

There are of course a few downsides:

- There is only support for one font size unless it is "baked into" a bitmap via the graphics buffer.
- There is not enough space to show the X, Y, Z coordinates, although I think it could be shown in place of the status string when nothing else needs to be shown there.
- Support for more than two nozzles would be difficult, but could probably be done by hiding the target temperatures.

Anyhow, this is just a proof of concept. Let me know if there is any interest in pursuing this.

-- Marcio

Roxy-3D, 2018-01-15:
There are a number of people that think the LCD code needs to be re-written.   It would be good to reduce the amount of data that needs to be transferred to the graphical display for each update.   The graphical LCD's are a big burden on the 8-bit AVR's.   And in fact, because the 32-bit processors can not talk the graphical LCD's at full speed, it slows them down too.

Being able to speed that up by just telling it where to display characters will be welcomed by a lot of people.  Especially the Delta people because they need every CPU cycle they can get when using an 8-bit AVR processor.

It would be nice if your proof of concept can be conditionally compiled in (and can cut out the UG8 code).   Then people that want to use it  (or are forced to use it because they need some of the CPU cycles back) can just turn it on and benefit from it.

CCS86, 2018-01-15:
Cool concept, I like it. 

I use this controller on my Ultimaker Original and it used to cause print hiccups when navigating the menu (Marlin 1.0). This seems to have gone away in 1.1, but I'm sure it still taxes the CPU. I personally don't care about the XY coordinate display, but Z would be nice.

On a side note, has anyone with this controller gotten proper 1:1 mapping of the rotary wheel to menu steps? I have tried a number of different "pulses per step" and "steps per menu item". All seem to leave me with a rounding error, so a click of the wheel is ignored, every so many clicks.

AnHardt, 2018-01-16:
@CCS86 
The pinstates for the rotary encoder are polled, so there is a good chance to miss a pulse.
Better, interrupt driven, ways depend on the interrupt capability of the pins. I don't know about any 8 bit board - display combination where the pins have interrupt capability.
But time will come, when we implement this for the 32 bit boards.

CCS86, 2018-01-16:
@AnHardt , that makes sense.

Any chance that ditching the bitmaps will free up more than enough overhead to increase the polling rate?

CCS86, 2018-01-18:
It doesn't seem like a polling issue, now that I think about it. If that was the case, you would just see a lag occasionally, before a new pin state was recognized, right? My issue definitely requires a new move of the encoder. It often seems to straddle two menu items, despite having solid detents.

thinkyhead, 2018-02-11:
I suggest this be implemented as a third "type" of LCD, in addition to "DOGM" and "HD44780". This will be much simpler than trying to patch up the DOGM code to handle this and the existing methodology.

marcio-ao, 2018-02-12:
@thinkyhead: I've implemented the "light" status screen on our branch of the LulzBot Marlin for our next gen printer, not so much for performance reasons, but because it looks better. As a first cut, all I really did was replace "lcd_implementation_status_screen" in "ultralcd_impl_DOGM.h" with my own, but the menu system continues to use u8g as it did before. There were a few minor modifications in other places, but it was a fairly well contained patch.

It's a bit peculiar to use native ST7920 calls for the status screen, and u8g for the menu, but it does allow the menus to continue to use the traditional look with a smaller font, which is nice as it allows five lines per screen as opposed to the maximum of four you would get with the ST7920 native font. The hybrid approach does not eliminate RAM or Flash memory used by u8g lib, but it does make the status screen more efficient CPU-wise as you are only updating characters rather than bitmaps. If there is an interest in this, I can look into merging it as an alternative status screen for CPU-constrained printers.

It is only half what I originally proposed, however, as it does not change the ultralcd menu at all and still requires u8g for that. Getting rid of that it entirely would be a lot more work, as the entire menu system would need to be re-implemented.

Anyhow, on a related note, I am now working on a color touch-screen interface for Marlin. Perhaps this could be put on the table as part of a Marlin UI overhaul?

AnHardt, 2018-02-12:
Another possibility would be to make a lightweight LiquidCristal library for ST7920. Implementing only the few functions we need. If this wouldn't use the 4 native (16x16) definable chars, but could implement 8 custom graphic characters (8x16), this could be used as a normal 16x4 display from 'ultralcd_impl_HD44780.h' (limited to ASCII (up to 127) with just some larger additional char graphics definitions).

marcio-ao, 2018-02-12:
@AnHardt : I am not familiar with the HD44780. I had assumed it was already using the ST7920 text mode with custom characters. Is this not the case? Is there a screenshot of what it looks like so I can get a sense for what it looks like?

thinkyhead, 2018-02-19:
[image]

marcio-ao, 2018-03-28:
Closing this as it has already been merged.

closed 2018-03-28 by marcio-ao