MarlinFirmware/Marlin #9192: Custom Boot Screen Error

Yurtkany, 2018-01-15:
Hi i am using marlin 1.1.8 version and Rumba Board . I tried to add custom boot screen but i receive error message.
exit status 1
'custom_start_bmp' was not declared in this scope
Can anybody help me ?

AnHardt, 2018-01-15:
example in ...\examples\BQ\Hephestos_2

Yurtkany, 2018-01-15:
Sorry i forgot write in issue . I modified _Bootscreen.h and copyed to marlin folder but i receive error

AnHardt, 2018-01-15:
Doublecheck your bootscreens array name. It must be 'custom_start_bmp', Any other name will fail.

Yurtkany, 2018-01-16:
i checked all is ok but its still receive error. can you write full instructions because i cant find in internet.

Yurtkany, 2018-01-16:
ok i did it but this time buzzer always beeping how can fix ?

closed 2018-01-16 by Yurtkany