MarlinFirmware/Marlin #9195: [FR] Autoload filament with filament presence sensor

KlausHans, 2018-01-15:
Hello,

sorry if this was already discussed but unfortunately i did not find anything in that direction.

I would like to have an autoload feature in combination with a simple switch based filament sensor. Currently it is only possible to do something if the filament runs out. I want to automatically load the filament for a given length, if the sensor is triggered by a new filament. 

My use case is a bowden extruder in an enclosured printer, where i can't get well to the spring loader. This feature would be really handy here.

Of course this should be only triggerable if no print is active and no filament runout was detected and the filament is being unloaded (this would trigger a filament load from the other side).

thinkyhead, 2018-02-11:
In most instances this would be undesirable. The issue is that if you have the type of sensor that most of us do, it may take a few seconds to thread through the filament sensor, then into the extruder, and finally get aligned with the drive gear. In that case you would not want the extruder to start spinning as soon as you triggered the sensor.

However, if you have a sensor that sits extremely close to the filament drive then it can work.

Developers here are quite busy working on the new release. But if you need this very much, you may be able to hire a developer for $$$ to implement the feature for you.

KlausHans, 2018-02-11:
"In most instances this would be undesirable. The issue is that if you have the type of sensor that most of us do, it may take a few seconds to thread through the filament sensor, then into the extruder, and finally get aligned with the drive gear. In that case you would not want the extruder to start spinning as soon as you triggered the sensor."
Honestly i don't understand. As sensor i thought of something like this: https://www.thingiverse.com/thing:2443425 Just simple presence.
This would be direct in before the extruder (you distinct between extruder and drive gear. Isn't the drive gear placed in the extruder?). From sensor to drive gear there would be some cm. Why would be this an issue?

The Prusa i3 MK3 has a similar feature, it looks so handy.

"Developers here are quite busy" I completely understand, its just a suggestion. Maybe some day i will have a look by myself.

thinkyhead, 2018-02-19:
> Honestly i don't understand. As sensor i thought of something like this: https://www.thingiverse.com/thing:2443425 Just simple presence.

The CR-10 filament sensor is 2-3 inches in front of the drive gear. See https://www.youtube.com/watch?v=as9iyR-slrc

lars-sorensen, 2020-04-27:
I think this would be an excellent addition!
I love the way it works on Prusa MK3S

SparkyDan555, 2020-07-23:
I agree that this would be a very helpful addition. The Prusa MK3S does this perfectly. I don't believe the position of the runout sensor matters since the full prime length falls short of the nozzle anyway, it is only the repeated "Purges" that make the filament reach the inside of the nozzle. So if the filament load started a few seconds early because of the gap between the sensor and feeder wheel, the user can simply keep pressing "Purge more" until filament is in the hotend. This should be fairly easy to implement since the runout sensor is simply serving the function of the LCD encoder button etc when Marlin asks the user to insert filament and press the button to start the filament prime. The sensor would instead start the process.

BsCmOD, 2020-08-17:
Follow, I'm interested to this feature implementation in Marlin. I'd tried to do it myself (as I've done for some simple functions) but I can't do it. It's could be a cool feature by set the function on or of into configuration.h anche in marlin menu, it could be also set a delay from the filament sensor detection to the load and also the length and the loading speed

BsCmOD, 2020-10-15:
UP, is there some hope to see this function implented into the next future in Marlin?

ntmaster26, 2020-10-24:
Bump ... no autoload ?

niccoreyes, 2020-11-15:
Finally starting to get used to the marlin syntax, just added SD_AUTOOPEN_MENU where it automatically reads the sd card when inserted like prusa firmware, I think I can make a few more mods for autoload feature. Although it may take a while when I'm not busy. I'm currently running on an MMU setup at the moment and I made a few pull requests recently for MMU improvements as well. This very much possible with one define statement

niccoreyes, 2020-11-15:
made a quick commit and branch, but I'm still not confident enough to make a pull request out of this
I haven't tested as I haven't reverted my MMU setup on my Marlin-ified Prusa printer
I have another prusa printer still running on stock firmware but I don't wanna touch that

https://github.com/niccoreyes/Marlin/tree/FILAMENT_AUTOLOAD

https://github.com/niccoreyes/Marlin/commit/73af75bf3359ce7236410a1aefd0871c56e5b5b4

BsCmOD, 2020-12-02:
> 
> 
> made a quick commit and branch, but I'm still not confident enough to make a pull request out of this
> I haven't tested as I haven't reverted my MMU setup on my Marlin-ified Prusa printer
> I have another prusa printer still running on stock firmware but I don't wanna touch that
> 
> https://github.com/niccoreyes/Marlin/tree/FILAMENT_AUTOLOAD
> 
> [niccoreyes@73af75b](https://github.com/niccoreyes/Marlin/commit/73af75bf3359ce7236410a1aefd0871c56e5b5b4)

Have you got some news?

thisiskeithb, 2021-09-23:
#22655 added `FILAMENT_CHANGE_RESUME_ON_INSERT` which should cover this FR:

https://github.com/MarlinFirmware/Marlin/blob/9c208a008d4f336f1bdc095af0a50754311092d1/Marlin/Configuration_adv.h#L2428

closed 2021-09-23 by thisiskeithb