MarlinFirmware/Marlin #9196: Setting up Dual Y motors on Ramps 1.4

donal1, 2018-01-15:
Newbie using ramps 1.4 and nema stepper motors running off 12 volts. I've been looking around for awhile and haven't found anything that gives a step by step guide of setting up dual y stepper motors running off E1. In my case I don't care if it runs off E0 or E1. 

My issue is that the stepper motor that should be running off the E0 or E1 doesn't run. I have changed the code in the configuration.adv.h to define dual y and then tried a number of combinations of Y2 changes in the pin.ramps. But nothing has fixed the problem it seem there is no current even running to the second motor. If anyone has a marlin firmware solution I'd be very appreciative as I don't want to solve this with soldering or through hardware.  

Extra info my board is also running two z axis and an x axis that works fine. It's just this second motor that doesn't work even if I swap out the cables or try different motors

Bob-the-Kuhn, 2018-01-16:
It sounds like you have the following setup:
X - single motor
Y - single motor
Z - two motors run off of one driver
E0 - the extruder
E1 - second Y motor

This is a fairly standard setup.

I take it the stepper driver works in other sockets and the motor (and it's cable) work when attached to a different socket.  If that's the case then it's either a configuration problem or the RAMPS board is bad.

Please ZIP up your two config files and your pins_RAMPS.h file and attach to this post.

donal1, 2018-01-16:
[Marlin zip.zip](https://github.com/MarlinFirmware/Marlin/files/1633471/Marlin.zip.zip)

donal1, 2018-01-16:
Yep the set up is the same as above and the cables do work. But anytime I attach the cables to the E1 or E0 there's no movement.

donal1, 2018-01-16:
Also to update I won't be using an extruder so I can use either E0 or E1.

revilor, 2018-01-16:
No guarantee, but from reviewing the source code I would say: revert pin_RAMPS.h to the original version and it should work. The first free extruder stepper is used as Y2 by default but due to your changes now points to pins which don't exist.

donal1, 2018-01-16:
Hey so I reverted the pins back to the original, E0 now Y2 and E1 left the same. But now I'm getting a compiler error. 

Speficially "E0_STEP_PIN.E0_DIR_PIN, or E0_ENABLE_PIN for this board". 

Let me know what you think.

revilor, 2018-01-17:
Looks like now the E0 pins are not defined at all. You need E0 and E1 pins back to their original values, no Y2 definitions are required in pins_RAMPS. The Y2 = E1 magic will happen in another place.

As you have defined EXTRUDERS 1 
E1 will be used as Y2

donal1, 2018-01-17:
Awesome, so if I just put E0 back in place it all should run. Man I hope this works. Thanks revilor sound.

donal1, 2018-01-17:
No luck, I've set the definitions back but no movement. 
[pins_RAMPS.zip](https://github.com/MarlinFirmware/Marlin/files/1638823/pins_RAMPS.zip)

Bob-the-Kuhn, 2018-01-17:
The last piece of the puzzle might be the line **#define INVERT_Y2_VS_Y_DIR true** in the file **Configuration_adv.h**.

If mechanically the two steppers need to rotate in the same direction then set this to **false**. 

If this were the problem then I'd expect an awful racket as the two steppers fought each other.

Which flavor of Marlin are you using?  I tried it on bugfix-1.1.x and I could see the Y & E1 socket signals behaving as expected.

If the above doesn't fix it then please do a sanity check. 
1. Remove all the cables and stepper drivers except for X.  
2. Command X movements.  X stepper should work.
3. Move the X cable and stepper driver to the Y socket.
4. Command Y movements.  X stepper should work.
5. Move the cable and stepper driver to the E1 socket.
6. Command Y movements.  X stepper should work.

____

You were on the right track when trying to change the pins_RAMPS.h file so that you could use the E0 socket for Y2.  If you change all the **Y2** to **E1** in the commented out section then it would work.

____

Curiosity question - since you're not using the extruder, what are you doing?

donal1, 2018-01-17:
I'm using the latest release (1.1.x.zip … 1.1.8), I guess I'll download the bugfix version asap.
The steppers will need to rotate in opposite directions.

I've try different combinations but to be sure run through it the way it described above. 
I tried changing the E1 to Y2 but still nothing (after I saw your good post on dual X on E0), then I reverted back to normal with E0 as Y2 and now I've reverted everything back to the original with E1 and E0 staying the same and still no movement. 

To your curiosity question top secret for the moment.

donal1, 2018-01-17:
Nope still not working, downloaded the bugfix version. I swapped out the cables as you describe still nothing. 

What could be the problem? :(

Bob-the-Kuhn, 2018-01-17:
Either there's a hardware problem or we have an apples/oranges problem.

Take bugfix-1.1.x and make only the following mods:
Configuration.h - set your baud rate
Configuration_adv.h - enable the Y dual steppers

Now try the E1 socket (connected to the X stepper) and see if it works.

If it doesn't work move the driver & cable to E0 and see if it works.

If it doesn't work then modify the pins_RAMPS.h file so that Marlin thinks the E0 socket is E1 and try it.  Here's the modified file.
[pins_RAMPS.zip](https://github.com/MarlinFirmware/Marlin/files/1640972/pins_RAMPS.zip)

____

If you have access to resistors and LEDs, try tying a resistor & LED from each of the following signals to ground.
pin 1 - ENABLE (lower left most pin in each socket when oriented as shown below).
pin 7 - STEP
pin 8 - DIRECTION

Start with the Y socket & verify that all three show activity when doing a +Y followed by a -Y movement.  

When you know the LEDs are setup correctly them move them to the socket you're using for Y2.  Do the +Y , -Y movement and see if all three behave the same as before.
 
[image]

----

I have double top secret clearance so you can tell me.  😉

donal1, 2018-01-18:
Sure I can try that I a couple of hours I currently using the motor for my printer. I'm just confused by a few things. 

When you say " try tying a resistor & LED from each of the following signals to ground." What exactly do you mean? 

Like attach an LEDs each of the pins of the sockets and when I send signals over pronterface it should light up? What resistor should I use? I get pin 1 how can I find pin 7 and 8? and this may seem like a very dumb question but how can I attach the led/resistor to ground? 

Only double top secret clearance, I need triple. If you fix this though you'll be at centuple clearance.

Bob-the-Kuhn, 2018-01-18:
I was hoping that you were handy with a soldering iron & could quickly put together a poor man's logic analyzer.

I have a bunch of female-female jumpers meant to be used on Arduino projects.  I just cut up a couple different colors, slid heat shrink tubing on each wire , soldered a 100 - 500 Ohm resistor to a LED, soldered that to the two wires, trimmed the extra lead wires, slid the heart shrink into place & heated the shrink tubing.

I also ripped apart some spare headers so I have pins that are long enough to ensure good contact between the LED connector and the stepper socket. 

The LEDs are more of a curiosity thing.  If you can't get Y2 working by playing with pins_RAMPS.h then you've got faulty hardware.  No other option but to buy a new mega2560 & RAMPS.  The LEDs might allow us to determine if it's the mega2560 or the RAMPS but ... it costs more to buy them separately so any info obtained from the LEDs won't affect what you do.

----

FYI - here's the pinout of the stepper drivers when looking down from above.

Be careful when inserting them.  Plugging one in backwards will fry your board.

```
            ----------
  /ENABLE  O|1     16|O  VMOT
      MS1  O|2     15|O  GND
      MS2  O|3     14|O  2B
      MS3  O|4     13|O  2A
   /RESET  O|5     12|O  1A
   /SLEEP  O|6     11|O  1B
     STEP  O|7     10|O  VDD
      DIR  O|8      9|O  GND
            ----------

```

donal1, 2018-01-18:
Still no luck, I think I'll forgo the building of the logic analyzer (I don't have the any solder or slip ties) and just buy a new mega and ramps board. I checked the pins with my multi-meter and the E pins were a pretty much zero voltages while the x and y were at 12v. Thanks for the help really great ideas.

Bob-the-Kuhn, 2018-01-18:
12V?  With the driver removed, only pin 16 in the socket should be at 12V.  That pin should be 12V on all five sockets.  All other pins should be either floating or 0-5V.

I've attached a picture with the pin 1 locations circled in blue.

[image]

[image]

donal1, 2018-01-18:
I was checking the x and y axis pins with drivers in place. It has to be hardware problem, I've checked and rechecked the motors, cables, drivers and different combinations of code with no avail. Honesty I think replacing the ramps and the mega is my only option at this point.

revilor, 2018-01-18:
What kind of stepper drivers are you using? When not one stepper works, maybe the drivers are just placed the wrong way around?

donal1, 2018-01-18:
I'm using a4988. No one stepper does work its the stepper attached to the extruder pins that doesn't. I've checked it the drivers are definitely the right way around.

Bob-the-Kuhn, 2018-01-19:
12V on the X, Y & Z sockets but not on the E0 & E1 sockets means the RAMPS card is bad.

donal1, 2018-01-19:
Damn bad board, I don't suppose theres a relatively cheap and easy way to fix it?

Bob-the-Kuhn, 2018-01-19:
Find someone with a soldering iron and, on the bottom side,  jumper pin 16 from any of the working sockets to pin 16 on any of the non-working sockets.

Just to see if this is the only issue, if you have a small enough solid wire you could insert it in pin 16 of a working socket and pin 16 of E0.  You should then have a fully functional E1 socket.

I suggest contacting the site where you bought your RAMPS card.  Most of the time they'll cheerfully send you a replacement.

$8 RAMPS shipped from US location.   https://www.ebay.com/itm/RAMPS-1-4-RepRap-Mega-Pololu-A4988-extend-Shield/201878106149?hash=item2f00df7025:g:gzoAAOSwB-1YqmFs

$10 RAMPs, 2560 & LCD shipped from China   https://www.ebay.com/itm/RAMPS-1-4-Mega2560-12864-LCD-A4988-Driver-Great-For-3D-Printer-Arduino-Reprap-S/322977799270?rt=nc&_trkparms=aid%3D222007%26algo%3DSIM.MBE%26ao%3D2%26asc%3D44039%26meid%3D59d8b5d5c7164350a00803ba13660e6a%26pid%3D100005%26rk%3D2%26rkt%3D6%26sd%3D142535302590&_trksid=p2047675.c100005.m1851

donal1, 2018-01-30:
So back to square one, actually I'm probably lower at one at this point. 

To catch back up, I tried to set up my ramps 1.4 running, with marlin and a4988 drivers with pronterface running from my laptop, 4 motors work except that the 5th motor that I want to run off the empty extruded port does not work. 

I'm fairly certain this isn't a software problem so I bought a new ramps and mega, (the mega has yet to arrive so I'm just trying out the new ramps). But now as I run the same code as before with the same drivers and motors NOTHING moves. All I get is noise and the motors heat up but doesn't rotate and when I feel the 2nd Y motor I can tell that no current is flowing into it (loose and can easily move it with my hands). 

This is insane, I'm really starting to run out of options so I'd really apprecaite any help at all.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1679346/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/1679349/Configuration_adv.zip)

Bob-the-Kuhn, 2018-02-01:
I've just ordered a RAMPs from that link I posted earlier.  I'll see if it does anything strange.

First I'll need to modify my printer to use a RAMPS system.  

I expect it'll be Tuesday or Wednesday before I have anything solid.

donal1, 2018-02-01:
Yeah, I tried my ramps again and the z axis does work there was a problem with my design of my rig. 

I thing I'm gonna give up on the dual y at this point and redesign the machine so it needs only one y axis. 

Bob I wanna say thanks you've really helped me out over and over again. If your still interested in the project private message me and I can give you the load down.

Bob-the-Kuhn, 2018-02-02:
What version of Marlin are you using?  

I just tried the files posted 2 days ago and got some "out of date" error messages.

Bob-the-Kuhn, 2018-02-02:
Don't give up yet.

bugfix-2.0.x has been broken for 2560 boards for about a week.  Last night's update still left the dual Y broken.

I haven't looked at bugfix-1.1.x yet.

revilor, 2018-02-02:
Before redesigning your machine wouldn't it make sense to - at least for the moment - drive the two Y motors from a single stepper driver just like it's done for the Z motors? RAMPS is not prepared for this with a second Y connector, but wiring two steppers in series or parallel is not that complicated.

Bob-the-Kuhn, 2018-02-03:
I've verified that the dual Y motor configs work in both bugfix-1.1.x and bugfix-2.0.x.  This was on a 2560 + RAMPS setup.

Attached are the two configurations I used.  I took your very first config and updated it to the latest .  There were just a few more options and some changes to the comments so your config was fairly recent.

I did change the baudrate to 250k.

[Marlin 1.1.x Config.zip](https://github.com/MarlinFirmware/Marlin/files/1691365/Marlin.1.1.x.Config.zip)
[Marlin 2.0.x Config.zip](https://github.com/MarlinFirmware/Marlin/files/1691367/Marlin.2.0.x.Config.zip)

thinkyhead, 2018-02-17:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

kd6hq, 2018-07-17:
I am wanting to set up a second Y stepper motor to use with a laser setup.
None of the firmware files have been modified yet (Merlin 1.1.8).
From reading thru this post it appears that the only thing I need to do is to 
#define Y_DUAL_STEPPER_DRIVERS  in Configuration_adv.h.  Is this correct?
thank you

thinkyhead, 2018-11-05:
Correct.

closed 2018-11-05 by thinkyhead

jgilbertfpv, 2018-12-02:
@kd6hq you are doing exactly what i'm currently trying to do with a laser and dual y motors. have you got your setup working? i've been looking at an older turnkeytyranny fork but i'd like like to use the newest stable release of marlin.

eyewy, 2018-12-13:
> Looks like now the E0 pins are not defined at all. You need E0 and E1 pins back to their original values, no Y2 definitions are required in pins_RAMPS. The Y2 = E1 magic will happen in another place.
> 
> As you have defined EXTRUDERS 1
> E1 will be used as Y2

@revilor, can you please explain where the "magic" happens? I want to use the second extruder driver to drive a 2nd motor for y axis. Thanks!