MarlinFirmware/Marlin #9198: G92 does not appear to work

fiveseven808, 2018-01-16:
------------------------------------------------------------
### Bug Report
- Description: 3D Printer has no endstops. Do not send G28. Manually place hot end at 0,0,0. Send G92 command to zero axises. If I had move axises via command, LCD or Gcode, the display shows a specific value for each axis, rather than 0,0,0.
- Expected behaviour: G92 will tell the printer that the Axises are at 0,0,0 allowing for a followup print. 
- Actual behaviour: Nothing happens, and when the printer resumes/starts another print, it runs into software endstops and ruins the print. 
- Steps to reproduce:
  - Move tool head via G1 or something else. 
  - Do a G92 command
  - Check display attached to printer to see X Y Z status. 

G92 via Gcode or sent over serial does not re-zero axises even though I do it by hand. This requires me to reset the printer every time I want to print. 

In addition to this, I was wondering if there was a way to control your Z axis to go below 0. I.e. I reset my printer with Z at Z_max and I'd rather not spin the threaded rods a million times to return to Z=0

tcm0116, 2018-01-16:
I'm guessing that an assumption was made that the printer would be homed to ensure that it is in a known condition. G92 then provides an offset to the workspace, but the printer continues to respect the physical limits via the softstops. You could try using M211 to disable the softstops.

fiveseven808, 2018-01-17:
I'm assuming so too... but since the Set position G92 command exists, wouldn't you think that you'd be able to manually set a home with it? That's how I would imagine it working at least... 

I'm hoping we can keep this issue open so that someone can make a flag that disables the "known condition" check and allows a manual condition set via G92 or otherwise. After all, I can print absolutely fine, so long as I reset my printer completely before doing so.

thinkyhead, 2018-01-20:
To get `G92` to behave like it did in Marlin 1.0 you can enable `NO_WORKSPACE_OFFSETS` and then it just sets the current position without any other effects.

michaelnew, 2018-06-13:
@fiveseven808 Are you able to confirm that `NO_WORKSPACE_OFFSETS` works for you? I would like to switch from Smoothieware to Marlin, but I don't use endstop switches either (in my case I set the X and Y position by detecting when the inductive sensor moves off the bed), and being able to set the position of the axes without having to do `G28` first is a must.

Smoothieware supports a `G28.3` command which, according to their docs: `allows you to set a home position manually without moving to limit switches`. I'd need the same functionality in Marlin to be able to use it.

closed 2018-06-13 by thinkyhead