MarlinFirmware/Marlin #9200: Move controller fan variable definition

ExtremeSitting, 2018-01-16:
I'm not sure how this should be categorized. It's more of a request to reorganize things.

The variable that enables the use of the controller fan is located in configuration_adv.h and has been since at least 1.0.1. However, it's not very obvious that fan pin 1 and 2 are not enabled or controlled by default. This may be a very specific issue, but on the Taz 5 and 6 fan pin 2 controls the fan that cools the steppers and the board as a whole. Since the Z axis is controlled by a single stepper, the fan is critically important. Adding to the issue, it's not obvious if the fan isn't running without close inspection. In my case, I didn't realize the fan wasn't running until it was too late. This was more than a year ago now and I have learned my lesson.

I am fully aware this was my mistake and does not represent a logical bug or problem with the firmware.

I think it would be immensely useful to move the "USE_CONTROLLER_FAN" definition to configuration.h around where the board type is defined and perhaps with a comment about setting the fan pin in the other file.  In the best case, this would save someone from burning out steppers or an entire board.

closed 2018-02-03 by ExtremeSitting

thinkyhead, 2018-02-05:
For most, I imagine the controller fan is connected to an always-on 12V plug. But for those who want to make it an add-on perhaps it would be easier to find it if it was kept in the main Configuration file.

AnHardt, 2018-02-05:
> But for those who want to make it an add-on perhaps it would be easier to find it if it was kept in the main Configuration file.

That's true for any option in Configuration_adv-h. We have to make a cut somewhere. However, the importance will always be a matter of personal taste/needs. Even probes, ABL, power supply, ... could be considered as advanced.

Can we find and agree about a rule upon that? I don't think so. For someone the decision will always be random.

Reading both files is mandatory for every user - and, of cause,  a 'basic' understanding of at least the options he/she wants to change. (Even the 'pros' have lost the overview about the side effects of options interactions, a long time ago. (We just have a better feeling for what could interact with each other))