MarlinFirmware/Marlin #9201: [CE] Filament width display fails to compile.

ExtremeSitting, 2018-01-16:
### Compile Error
When I compile with `SDSUPPORT` and `FILAMENT_LCD_DISPLAY` I get an error:
```
Arduino: 1.8.3 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\ultralcd.cpp:96:0:

sketch\ultralcd_impl_DOGM.h: In function 'void lcd_implementation_status_screen()':

ultralcd_impl_DOGM.h:656: error: 'wstring' was not declared in this scope

       strcpy(wstring, ftostr12ns(filament_width_meas));

              ^

ultralcd_impl_DOGM.h:657: error: 'mstring' was not declared in this scope

       strcpy(mstring, itostr3(100.0 * (

              ^

In file included from sketch\ultralcd.cpp:96:0:

ultralcd_impl_DOGM.h:748: error: 'wstring' was not declared in this scope

         lcd_print(wstring);

                   ^

ultralcd_impl_DOGM.h:751: error: 'mstring' was not declared in this scope

         lcd_print(mstring);

                   ^

exit status 1
'wstring' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

This seems to be simply due to `SDSUPPORT` being enabled which causes [line 646](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/ultralcd_impl_DOGM.h#L646) of `ultralcd_impl_DOGM.h` to evaluate false, but further down it checks if both are enabled and still attempts to use `wstring` and `mstring`.

empakoso, 2018-01-19:
Just installed today’s 1.1.x bugfix and can not seem to get the filament width to display on the lcd at.
Just posting this here in hopes that this might be related to the same problem.  

It started to happen after I installed the 1.1.x bugfix dated jan11.

M407 reports but no LCD real-time readings.

ExtremeSitting, 2018-01-19:
@empakoso The variable that enables it is [here](https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/Configuration_adv.h#L1367). Do you have that enabled and was it able to compile with SD support?

Thanks for your comment.

empakoso, 2018-01-19:
Indeed, this variable you linked to is enabled.  I double checked it and uploaded the fw again just to be sure.  In fact that was the reason for loading yesterday’s bugfix, just to be absolutely sure.

I have experienced that scrolling long messages has prevented this from displaying in the past, so I have stopped using that.

I was able to compile with SD support.  I did read that there was some work being done on the SD code.  Could it be that you are using either copied over config files or outdated IDE prog/libraries?  I updated all of this before trying the latest bugfix.

empakoso, 2018-01-19:
I just realized that you are using a graphic display.   I am using the character display.  That could be the difference.

ExtremeSitting, 2018-01-19:
I updated my compiler and all the libraries before my last update. I think I'm going to re-pull and start from scratch. I'm not entirely sure that I used bugfix.

As of this moment, I have filament width control working fine, but nothing about it on the display. SD support is a bit more important recently. It sure would be nice to have both, but I'm not willing to edit the conditional to allow it without feed back from the maintainer. I'd like to know why it's necessary to have one or the other (FWD or SD support).

thinkyhead, 2018-02-14:
Patched. Have a nice day!

commit e2fef4df (2018-02-14): Fix #9201

FILAMENT_LCD_DISPLAY and SDSUPPORT are a-ok together.
commit e7b50c89 (2018-02-14): Fix #9201

FILAMENT_LCD_DISPLAY can work with SDSUPPORT
closed 2018-02-14 by thinkyhead