MarlinFirmware/Marlin #9207: [Question] Step Rates

CCS86, 2018-01-16:
Hi guys,

I am running 20 tooth GT2 pulleys on my UMO, with 1.8* steppers and was configured at 80 steps/mm (X&Y) and 16 microstepping. I recently swapped out to steppers with 0.9* step angle. Instead of doubling the steps/mm, I switched to 8 microstepping, in hopes of higher torque and better positional accuracy. I did subsequently change to non-integer steps/mm (79.5, 79.6) to fine tune sizing, which I read isn't an issue.

I have noticed an odd resonance in Y axis only, at moderate speeds (20-60mm/s).  I'm not sure if that is related to the 8 microstepping or not. I am considering bumping it back to 16 and doubling the steps/mm. Will this put me over the maximum step rate of the ATMega2560? The bold entry in the table here suggests maybe (but also says the table is out of date and the limits are too high):

http://reprap.org/wiki/Step_rates

If my 20 tooth pulleys, on a 0.9* stepper @ 16 microstepping gives me a maximum Print speed of 100mm/s, that is fine. I really don't print that fast.

Thoughts?

On a similar note, I have a question about Z stepping resolution. Occasionally, I do spiral vase prints. Looking at Cura's gcode output, the Z movements are in increments of 0.001mm. My Z drive is 200 steps/mm (and 16 microstepping). 200 * 0.001 = 0.2 steps... but that is micro steps right? How does Marlin handle moves that are less than 1 microstep? Does it accumulate them until they total a full microstep?

Thanks!!

Sebastianv650, 2018-01-16:
It's not possible to give a maximum possible step rate, it depends on way too much factors. All we can speak about are the limits where Marlin will switch to double and quad stepping:
0-10kHz: single steps
10-20kHz: double steps
>20kHz: quad stepping

I prefer to stay within single step range for print moves, for example if you want to print at 80mm/s your maximum steps/mm value is allowed to be <125. 125*80 = 10kHz.

Second question:
Don't mix up microstepping and Marlin steps. Yes, 200steps/mm * 0.001mm are 0.2 steps. This means, Marlin will do nothing in Z. Marlin always calculates the target position in steps, the needed steps along an axis is target-last position. So as long as target_mm * [steps/mm] = last_position, nothing will happen.

CCS86, 2018-01-16:
Awesome, thank you for the insight @Sebastianv650! 

When calculating step rate limitations (or even steps/mm) for Marlin, we are talking about microsteps though, yes? Otherwise, you could change microstepping, without adjusting steps/mm.

Do you think 8 microstepping could be the cause of that resonance? I've been changing a bunch of stuff lately, so it's hard to say for sure. Mechanically, it is in really good shape. I have gantry friction down very low and even.

CCS86, 2018-01-16:
Well, apparently I got ahead of myself. I have not flashed the new settings to drop to 8 microstepping. The machine is still at 16 microstepping and around 160 steps/mm (interestingly X is at 159 and the resonating axis Y is at 159.2).

Sebastianv650, 2018-01-17:
> When calculating step rate limitations (or even steps/mm) for Marlin, we are talking about microsteps though, yes? Otherwise, you could change microstepping, without adjusting steps/mm.

In fact microstepping is only relevant for the stepper driver. For Marlin, it's all about "steps". If Marlin has to execute 10 steps, it's not important if the stepper driver sends signals for full steps, microsteps or whatever. It just has to trigger 10 steps at proper steprate.
As you already stated, for the final result the important one is that steps/mm setting is adjusted as needed. If you need 10 full steps to get from point A to B, after switching to 16x microstepping on the stepper driver this will become 160 steps.

CCS86, 2018-01-17:
Okay, good, that was my understanding.

Any thoughts on non-integer steps/mm causing resonance? I never noticed it at the beginning of a print, only after it had been running for a bit. Loudest on Y only moves, still there on compound moves, quiet on X only. I'll keep my ear out for it and try setting Y axis to an integer steps/mm if I hear it again.

Thanks!

Sebastianv650, 2018-01-17:
No, step/mm is not related to resonance in terms of float or integer. Resonance is a function of mass and frequency. If if you compare noise at a fixed feed rate changing steps/mm, you are actualy changing the feed rate so you can get a difference.

GMagician, 2018-01-17:
@Sebastianv650 that interesting....
I have 32 microsteps and a 0.9°/step motor hence I have 400 steps x unit.
Max speed is 120mm/s that means 48KHz and it seems working...
This frequency is far from values you write...maybe I badly translated your above post

Sebastianv650, 2018-01-17:
I'm sure what you mean. There was a Markdown format error in my first post, maybe you read 20kHz as max frequency? It's not, above 20kHz Marlin switched to quad stepping. I corrected the format error.

There is no fixed upper limit in Marlin (OK there is one, but you will most likely never reach it..). At some point your CPU load will be 100%, meaning you get no higher speed even if you double the feed rate and you might see some fancy errors showing up like bad heater handling and so on.

GMagician, 2018-01-17:
Ok that was what happened when speed was higher than 120...
Thanks for your reply

closed 2018-02-11 by thinkyhead