MarlinFirmware/Marlin #9209: [2.0.x] "SD init fail" "Error:volume.init failed" on Creality Ender 2

CReimer, 2018-01-16:
### Bug Report
I just flashed my new Creality Ender 2 with Marlin 2.0.x to get rid of the GPL violating firmware. Unfortunately it can no longer read from SD cards.

I think this is somewhat related to #7467 and #7468. But everything mentioned there doesnt' help.
I tried all delay settings mentioned and also experimented with SPI speeds.

When I push "SD init" in the OSD or send M21 I can see a black bar on the screen with a scrolling vertical line from left to right.

I tried disabling MINIPANEL but this doesn't solve anything either.

On the serial console I see "Error:volume.init failed" and sometimes "SD init fail".

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1636582/Configuration.zip)

Thanks

Christopher

CReimer, 2018-01-16:
Switched from my configs to the example configs in bugfix-2.0.x branch and the bugfix-1.1.x branch.
With bugfix-1.1.x I can access the SD card after several tries. Haven't tried printing just yet but after the sd card is initialized it seems pretty stable.

With bugfix-2.0.x it's impossible to initialize the SD card.

morsyfadl, 2018-01-17:
SPI speed maybe? Try 1/8 speed

CReimer, 2018-01-17:
Unfortunately no. 1/8 speed doesn't help with bugfix-2.0.x and doesn't fix the sporadically working SD card with bugfix-1.0.x. Did a whole print with bugfix-1.0.x today. It took 3 or 4 tries to detect the SD card, but afterwards the print went fine.

Bob-the-Kuhn, 2018-01-18:
Maybe the card doesn't have an SD detect signal.

Comment out all instances of "#define SD_DETECT_PIN" in the pins_YOUR_BOARD.h file.

CReimer, 2018-01-18:
Nope. After following a few includes (pins_CREALITY_ENDER.h -> pins_SANGUINOLOLU_12.h -> pins_SANGUINOLOLU_11.h) I could only find one SD_DETECT_PIN which was already set to -1. Commenting this line didn't change the overall behaviour.

Something I came up with yesterday is, that the SD card gets always detected, when I insert the card and click the initialize SD button in the main menu immediately afterwards.

CReimer, 2018-01-18:
I tried #9109 and it didn't help at first. After I figured out how SD cards are handled in Marlin I started adding some debug log messages. As it turns out there are two different chip select pins defined in Marlin. LCD_SDSS and SDSS

LCD_SDSS is set to Pin 28 and SDSS is set to Pin 31. To cut this short pin 28 is wrong and pin 31 is right. But LCD_SDSS is prefered.

It still doesn't work right after powering on, but it works while switching sd cards and after removing the SD card once after powerup.

Changing this pin in bugfix-2.0.x didn't help. It still only works with bugfix-1.1.x. But at least I have a more or less fully working printer again.

thinkyhead, 2018-02-02:
> LCD_SDSS is set to Pin 28 and SDSS is set to Pin 31. To cut this short pin 28 is wrong and pin 31 is right. But LCD_SDSS is prefered.

In your case. We'll need more corroboration before assuming it applies universally.

thinkyhead, 2018-02-02:
Is it possible that our `BOARD_CREALITY_ENDER` should just be dropped? The `pins_MELZI_CREALITY.h` file has the correct pin assignment for SDSS. And maybe the other pins are also more correct.

CReimer, 2018-02-02:
The `BOARD_MELZI_CREALITY` does work with bugfix-1.1.x and I don't have to edit any pins files. Unfortunately it's still broken with bugfix-2.0.x.

> Could this really be wrong for all Sanguinololu 1.1 generic character displays? Or, does this only apply to the Ender setup?

I'm not sure. There could be multiple revisions with different layouts. But I can't find any revision markings on my board. I could disconnect everything and upload some hi-res pictures of it, if you want those.

thinkyhead, 2018-02-16:
> The BOARD_MELZI_CREALITY does work with bugfix-1.1.x and I don't have to edit any pins files. Unfortunately it's still broken with bugfix-2.0.x.

The pins should match. If not, that will be corrected shortly.

If the pins already match then the issue is not the pins file, but something specific to the 2.0 AVR HAL.

thinkyhead, 2018-02-18:
@CReimer — There have been some minor tweaks and patches to `bugfix-2.0.x` today. See if it will work. If you find it still doesn't work, let's see if we can hunt down the cause. When it fails, does it just reboot the board over and over, or does it do something worse?

thinkyhead, 2018-02-26:
Duplicate of #8746

Let's move discussion to that issue.

closed 2018-02-26 by thinkyhead