MarlinFirmware/Marlin #9210: TMC2130 QUICK_HOME and sensorless endstop

mamama1, 2018-01-16:
In current Marlin 1.1.8, when using the TMC2130 and sensorless homing, enabling QUICK_HOME breaks sensorless homing completely so that the end of the printer is not detected.

quite noisy... :-)

Artem-B, 2018-01-31:
Yup. Marlin does not enable stallguard and completely ignores the stall.
As a workaround you can try enabling `#define HOME_Y_BEFORE_X` in Configuration_adv.h.
This should avoid quick homing and should keep sensorless homing working.

thinkyhead, 2018-02-14:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

boelle, 2019-02-20:
@mamama1 Please test with the latest  `bugfix-2.0.x`
1.1.8 has been put in the grave long time ago

boelle, 2019-03-06:
@thinkyhead i think we can close this one

mamama1, 2019-03-06:
i will report if there continue to be issues with 1.1.9 or bugfix 2.0.x

closed 2019-03-06 by mamama1

boelle, 2019-03-06:
we are mostly interested in bugfix 2.0 :-D

mamama1, 2019-03-06:
i might have it on my AM8 with the sensorless homing however I've done so little 3d printing in the past (and if, it was hassle-free), so that I don't even know what firmware or firmware settings i have on my printers right now :-D