MarlinFirmware/Marlin #9211: Having all TMC2130 on RAMPS or MKS GEN 1.4 with Dual Z drivers

mamama1, 2018-01-16:
Hi

- RAMPS or MKS GEN 1.4
- X, Y, Z, Z2 and E0 are all TMC2130 in SPI mode
- #define Z_DUAL_STEPPER_DRIVERS is enabled

When using Z_DUAL_STEPPER_DRIVERS, the next free Extuder is used as Z2. This does not work for RAMPS and all TMC2130 in SPI mode. Compilation error is that Z2_CS_PIN is not defined as this is missing in pins_RAMPS.h and also it seems that this is not remapped to E1 or whatever the next free Extruder is.

i simply added 

```
#define Z2_CS_PIN          36
#define Z2_CS_PIN          34
#define Z2_CS_PIN          30
#define Z2_CS_PIN          44
```

to pins_RAMPS.h and now I get a whole lot of redefine warnings but Marlin compiles.

teemuatlut, 2018-01-17:
Settle on a one pin that you use for Z2 CS and define it once only.

mamama1, 2018-01-17:
lol I somehow overlooked that I was redefining Z2_CS_PIN in the snippet I submitted with the issue xD

however, still the automatic remapping does not work like if the second Z driver was an ordinary step/dir driver.

Lyr3x, 2018-01-20:
You are not using Z2 with MKS Gen 1.4.. ran into that issue before. You need to define Z1, E0, and E1 (this is your Z2).

mamama1, 2018-01-20:
@Lyr3x the sanity check was complaining that Z2_CS_PIN was not defined, so I defined it...
E1 is already defined in the pins_RAMPS.h.

Lyr3x, 2018-01-21:
May you post your configuration_adv.h?

sivonxay, 2018-01-23:
Hi, i'm having this same issue, except it's with a dual Y config. After uncommenting #define Y_DUAL_STEPPER_DRIVERS, I got an error in the sanity check saying that Y2_CS_PIN was undefined

So I defined Y2_CS_PIN to the pin the driver is plugged into, compiled and uploaded the firmware, but there was no motion in the second y axis.

decor82, 2018-02-28:
Hello,

i have the same error. I tested the 1.1.8 release and the bugfix-1.1.x from yesterday (02.27.2018) and i am not able to use dual Z drivers with the TMC2130 drivers.

Error message from arduino.

stepper_indirection.cpp:154: error: 'Z2_CS_PIN' was not declared in this scope

     _TMC2130_DEFINE(Z2);

                     ^

sketch\stepper_indirection.cpp:135:104: note: in definition of macro '_TMC2130_DEFINE'

   #define _TMC2130_DEFINE(ST) TMC2130Stepper stepper##ST(ST##_ENABLE_PIN, ST##_DIR_PIN, ST##_STEP_PIN, ST##_CS_PIN)

                                                                                                        ^

exit status 1
'Z2_CS_PIN' was not declared in this scope

teemuatlut, 2018-02-28:
Add the CS pin definition into your `pins_***.h` file.
Marlin doesn't at the moment handle setting Z2 CS pins from free E like it does with DIR/EN/STEP.

decor82, 2018-02-28:
Thanks teemuatlut. 

I did not test this, because the comment from sivonxay from 01.23.2018 stated that even if the sketch is compiled with "#define Z2_CS_PIN" or "#define Y2_CS_PIN" the secound motor not moved.

I added the line "#define Z2_CS_PIN 44" to the pins_RAMPS.h file and could compile 1.1.8 without errors. Will test it later today.

Is it correct that i dont need to add #define Z2_STEP_PIN, #define Z2_DIR_PIN and #define Z2_ENABLE_PIN because Marlin uses it from E1?

teemuatlut, 2018-02-28:
> Is it correct that i dont need to add #define Z2_STEP_PIN, #define Z2_DIR_PIN and #define Z2_ENABLE_PIN because Marlin uses it from E1?

That is correct.

decor82, 2018-03-01:
I have tested the 1.1.8 Firmware with "#define Z2_CS_PIN" and it is working perfectly! 

Thanks :)

thinkyhead, 2018-03-11:
I've added patches for the bugfix branches to auto-assign the CS pins to X2, Y2, and Z2 in the same manner as the step, dir, enable pins, so it should no longer be necessary to add them manually.

closed 2018-03-11 by thinkyhead

Neiljclements, 2018-10-10:
I got a bit lost readng this.  I have just wired my MKS Gen L with TMC2130's for X, Y, Z1 & Z2.  I have an LV8729 for the Extruder (E0).  Because I'm not using a 2130 on the extruder, I connected the CS from the second Z stepper to where the exrtuder (E0) would've connected.  IS this correct?  Or should I connect the second Z stepper somewhere else on the MKS Gen L?  
I'm quite new to all of this, so if you wouldn't mind breaking down exactly what I need to do please.
Thank you very much.

thinkyhead, 2018-10-10:
- Connect your E0 stepper motor to the first E plug on the board (E0).
- Connect your Z2 stepper motor to the second E plug on the board (E1).
- Connect the Chip Select line for Z2 (E1) to the `E1_CS_PIN`.

Neiljclements, 2018-10-10:
Sorry for being a bit of a noob, but which is my E1_CS_PIN? I've looked at the MKS Gen L woring diagram and I ca't locate it.

PR #10057 (2018-03-11): [2.0.x] Auto-assign CS pins for X2, Y2, Z2
Fixes #9211 

Auto-assign CS pins for X2, Y2, and Z2 axes in the same manner as other stepper driver pins.
commit d385b9aa (2018-03-11): Auto-assign CS pins for X2, Y2, Z2 (#10057)

Fixes #9211
PR #10056 (2018-03-11): [1.1.x] Auto-assign CS pins for X2, Y2, Z2
Fixes #9211 

Auto-assign CS pins for X2, Y2, and Z2 axes in the same manner as other stepper driver pins.
commit 879b54a8 (2018-03-11): Auto-assign CS pins for X2, Y2, Z2 (#10056)

Fixes #9211