MarlinFirmware/Marlin #9214: [2.0.x] Compile fails with UBL enabled (and Travis is broken)

benlye, 2018-01-17:
### Compile Error
When I compile with `AUTO_BED_LEVELING_UBL` I get the errors below.  

It seems to have been broken by https://github.com/MarlinFirmware/Marlin/commit/acb1e313746c721c470e8e045bdd98446b4c58c7 because MESH_MAX_X and MESH_MAX_Y are no longer being defined in Conditionals_post.h.

This is also causing Travis to fail.

```
In file included from Marlin\src\inc/MarlinConfig.h:33:0,
from Marlin\src\Marlin.h:25,
from Marlin\src\Marlin.cpp:31:
Marlin\src\inc/Conditionals_post.h:1086:26: error: '_MESH_MAX_X' was not declared in this scope
#define MESH_MAX_X _MESH_MAX_X
^
Marlin\src\feature/bedlevel/ubl/ubl.h:65:28: note: in expansion of macro 'MESH_MAX_X'
#define MESH_X_DIST (float(MESH_MAX_X - (MESH_MIN_X)) / float(GRID_MAX_POINTS_X - 1))
^
Marlin\src\feature/bedlevel/ubl/ubl.h:137:50: note: in expansion of macro 'MESH_X_DIST'
MESH_MIN_X +  0 * (MESH_X_DIST), MESH_MIN_X +  1 * (MESH_X_DIST),
^
Marlin\src\inc/Conditionals_post.h:1086:26: error: '_MESH_MAX_X' was not declared in this scope
#define MESH_MAX_X _MESH_MAX_X
^
Marlin\src\feature/bedlevel/ubl/ubl.h:65:28: note: in expansion of macro 'MESH_MAX_X'
#define MESH_X_DIST (float(MESH_MAX_X - (MESH_MIN_X)) / float(GRID_MAX_POINTS_X - 1))
^
Marlin\src\feature/bedlevel/ubl/ubl.h:137:83: note: in expansion of macro 'MESH_X_DIST'
MESH_MIN_X +  0 * (MESH_X_DIST), MESH_MIN_X +  1 * (MESH_X_DIST),
^
Marlin\src\inc/Conditionals_post.h:1086:26: error: '_MESH_MAX_X' was not declared in this scope
#define MESH_MAX_X _MESH_MAX_X
^
Marlin\src\feature/bedlevel/ubl/ubl.h:65:28: note: in expansion of macro 'MESH_MAX_X'
#define MESH_X_DIST (float(MESH_MAX_X - (MESH_MIN_X)) / float(GRID_MAX_POINTS_X - 1))
^
Marlin\src\feature/bedlevel/ubl/ubl.h:138:50: note: in expansion of macro 'MESH_X_DIST'
MESH_MIN_X +  2 * (MESH_X_DIST), MESH_MIN_X +  3 * (MESH_X_DIST),
^
Marlin\src\inc/Conditionals_post.h:1086:26: error: '_MESH_MAX_X' was not declared in this scope
#define MESH_MAX_X _MESH_MAX_X
^
Marlin\src\feature/bedlevel/ubl/ubl.h:65:28: note: in expansion of macro 'MESH_MAX_X'
#define MESH_X_DIST (float(MESH_MAX_X - (MESH_MIN_X)) / float(GRID_MAX_POINTS_X - 1))
^
Marlin\src\feature/bedlevel/ubl/ubl.h:138:83: note: in expansion of macro 'MESH_X_DIST'
MESH_MIN_X +  2 * (MESH_X_DIST), MESH_MIN_X +  3 * (MESH_X_DIST),
^
Marlin\src\inc/Conditionals_post.h:1086:26: error: '_MESH_MAX_X' was not declared in this scope
#define MESH_MAX_X _MESH_MAX_X
^
Marlin\src\feature/bedlevel/ubl/ubl.h:65:28: note: in expansion of macro 'MESH_MAX_X'
#define MESH_X_DIST (float(MESH_MAX_X - (MESH_MIN_X)) / float(GRID_MAX_POINTS_X - 1))
^
Marlin\src\feature/bedlevel/ubl/ubl.h:139:50: note: in expansion of macro 'MESH_X_DIST'
MESH_MIN_X +  4 * (MESH_X_DIST), MESH_MIN_X +  5 * (MESH_X_DIST),
```

benlye, 2018-01-17:
Should this:
```
    #if ENABLED(AUTO_BED_LEVELING_UBL)
      #define _MESH_MIN_X (max(X_MIN_BED + MESH_INSET, X_MIN_POS))  // UBL is careful not to probe off the bed.  It does not
      #define _MESH_MIN_Y (max(Y_MIN_BED + MESH_INSET, Y_MIN_POS))  // need *_PROBE_OFFSET_FROM_EXTRUDER in the mesh dimensions
    #else
      #define _MESH_MIN_X (max(X_MIN_BED + MESH_INSET, X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER))
      #define _MESH_MIN_Y (max(Y_MIN_BED + MESH_INSET, Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))
    #endif
  #endif
```

be this?
```
    #if ENABLED(AUTO_BED_LEVELING_UBL)
      #define _MESH_MIN_X (max(X_MIN_BED + MESH_INSET, X_MIN_POS))  // UBL is careful not to probe off the bed.  It does not
      #define _MESH_MIN_Y (max(Y_MIN_BED + MESH_INSET, Y_MIN_POS))  // need *_PROBE_OFFSET_FROM_EXTRUDER in the mesh dimensions
      #define _MESH_MAX_X (min(X_MAX_BED - (MESH_INSET), X_MAX_POS))
      #define _MESH_MAX_Y (min(Y_MAX_BED - (MESH_INSET), Y_MAX_POS))
    #else
      #define _MESH_MIN_X (max(X_MIN_BED + MESH_INSET, X_MIN_POS + X_PROBE_OFFSET_FROM_EXTRUDER))
      #define _MESH_MIN_Y (max(Y_MIN_BED + MESH_INSET, Y_MIN_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))
      #define _MESH_MAX_X (min(X_MAX_BED - (MESH_INSET), X_MAX_POS + X_PROBE_OFFSET_FROM_EXTRUDER))
      #define _MESH_MAX_Y (min(Y_MAX_BED - (MESH_INSET), Y_MAX_POS + Y_PROBE_OFFSET_FROM_EXTRUDER))
    #endif
```

Roxy-3D, 2018-01-17:
Yes...    Sorry about that!   

Travis was frozen last night....   It never finished...

I believe things are correct now.

THANK YOU!

benlye, 2018-01-17:
Yes, all good now - thanks.

Fixed by 9d0cf02fef695cdbfdcdbff133968d8508046cd9.

closed 2018-01-17 by benlye