MarlinFirmware/Marlin #9217: Ultipanel blinking and menu not working

lonnie7, 2018-01-17:
Marlin 1.1.8 
Arduino 1.8.5
MendelMax 1.5 (I know...Dinosaur)

I have it printing but the Display Blinks when xyz is showing zero. I like leaving the usb cable connected with power off but it beeps or blinks and makes a noise that bothers me. How can I stop the blinking? The speaker is;
//#define SPEAKER
//#define LCD_FEEDBACK_FREQUENCY_DURATION_MS 100
//#define LCD_FEEDBACK_FREQUENCY_HZ 1000

When I turn the rotary knob on the display to select the menu I have to turn the rotary 2 positions instead of one to see the menu and then I can not access anything in the menu. Pushing the rotary knob in does nothing.

I have;
#define LCD_LANGUAGE en
#define DISPLAY_CHARSET_HD44780 JAPANESE
#define ULTRA_LCD   // Character based
#define ENCODER_PULSES_PER_STEP 1
#define ENCODER_STEPS_PER_MENU_ITEM 5
#define ULTIPANEL 
 
I normally do not use the display to make changes but if it is there then I want it to work. But the most important of all..... how do I stop the blinking

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

lonnie7, 2018-01-19:
No Help?

AnHardt, 2018-01-19:
Just home the machine.

lonnie7, 2018-01-19:
That works when the power is on but not when the power is off. The USB still powers the display and I want to keep the usb plugged in without the blinking.

AnHardt, 2018-01-20:
```
// Warn on display about possibly reduced accuracy
//#define DISABLE_REDUCED_ACCURACY_WARNING
```

lonnie7, 2018-01-20:
I enabled the #define  DISABLE_REDUCED_ACCURACY_WARNING and uploaded but it still makes the blinking noise.
I was thinking it had to do with;
// Before homing the axis letters are blinking 'X' <-> '?'.
  // When axis is homed but axis_known_position is false the axis letters are blinking 'X' <-> ' '.
  // When everything is ok you see a constant 'X'

but I don't know what to change.

Thanks for the help

AnHardt, 2018-01-20:
..._adv.h
```
//#define HOME_AFTER_DEACTIVATE  // Require rehoming after steppers are deactivated
```
?

lonnie7, 2018-01-20:
nope. I get an alarm.

lonnie7, 2018-01-26:
No answer for the problem so I will go back to the old Marlin version and live with it.

thinkyhead, 2018-02-14:
It's not hard to modify the code:
```cpp
FORCE_INLINE void _draw_axis_label(const AxisEnum axis, const char* const pstr, const bool blink) {
  if (blink)
    lcd_printPGM(pstr);
  else {
    if (!axis_homed[axis])
      u8g.print('?');
    else {
      #if DISABLED(HOME_AFTER_DEACTIVATE) && DISABLED(DISABLE_REDUCED_ACCURACY_WARNING)
        if (!axis_known_position[axis])
          u8g.print(' ');
        else
      #endif
          lcd_printPGM(pstr);
    }
  }
}
```
I bet I don't even have to tell you what to change here.

closed 2018-02-14 by thinkyhead