MarlinFirmware/Marlin #9218: [bugfix-1.1.x] Motors are clacking and clicking after G29. Without G29 they are silent

ikarisan, 2018-01-17:
I am using a MKS Gen1.4 board together with three TMC2130 drivers for X, Y and E.  

Every time I print, the motors are starting to clack all the time. First I thought I am to narrow to the bed but **even without filament** it clicks and clacks during the print. Now I am sure that not the extruder motor rather the G29 (Bilinear bed leveling) causes this problem!

Use the attached GCODE files to comprehend:

[clicktest.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/1640490/clicktest.gcode.txt)
[clicktest_withG29.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/1640492/clicktest_withG29.gcode.txt)

The first GCODE file (clicktest.gcode) simply homes all axis and moves around the extruder the bed. You must have at least a 200x200 bed (or change the coordinates!). These movements are very silent, as expected!

The second GCODE file (clicktest_withG29.gcode) homes all axis and runs an auto leveling process (G29). These movements are filled with clicking and clacking all the time. 

The same happens if I recall the last leveling data with "M420 S1"!

And this is reproducible every time!

I made two YT videos. The first one shows the movements without G29 and the second one after G29.

Silent moves: https://youtu.be/TcR7T8kjJM4
Clicking moves: https://youtu.be/sVny98SFDNc

Here is another video, when I thought the extruder was the problem:
https://www.youtube.com/watch?v=LmIRYhW3xMk

thinkyhead, 2018-02-11:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

VanessaE, 2018-02-23:
@thinkyhead I wonder if this is the same issue we talked about elsewhere, where low acceleration causes stuttering when bed leveling is enabled?

thinkyhead, 2018-02-23:
The original post doesn't tell us very much. But if they're using UBL then this issue might be resolved by disabling `SEGMENT_LEVELED_MOVES`.

VanessaE, 2018-02-23:
I dunno what he's using, but I can confirm that disabling segmented moves fixes the majority of the stuttering for bilinear leveling, though straight moves are still being bisected, regardless of direction, length, starting position, or axis.

thinkyhead, 2018-02-23:
> straight moves are still being bisected

A single move from X0 to X100 will stop in the middle?

VanessaE, 2018-02-23:
Something like that, yes.  It accelerates from its starting point, moves to X=50, decelerating as it gets there, immediately accelerates again, and continues to X=100, as if I had issued `G1 X0 / G1 X50 / G1 X100`.  Even happens during G29 as it moves from point to point.

I'm having trouble reproducing that effect reliably, though.

thinkyhead, 2018-02-23:
Are you seeing the same issue I describe in #9360 ?

VanessaE, 2018-02-23:
With the g-code you presented there, the first four sides of the square move fine, the next three sides are "drawn" with two distinct segments apiece, the fourth of that set is done in one move, then the last four sides have one long move and one short move each.

thinkyhead, 2018-02-23:
Try adding `M420 S1` after the `G28` and see if it worsens.

VanessaE, 2018-02-23:
Ultimately, a little worse - more of the moves are segmented then.

thinkyhead, 2018-02-23:
The last move in any set of moves is always detached and done as a separate move.

VanessaE, 2018-02-23:
Now, if I load the kfactor gcode linked earlier, add in that `M420 S1` right after home, I don't see any obvious difference in how it prints, but as I'm using Sebastianv650 fork at the moment, and `SEGMENT_LEVELED_MOVES` is enabled there, I can't say if it should behave differently or not.

thinkyhead, 2018-02-23:
The Kfactor code probably has no `M400`s in it.

VanessaE, 2018-02-23:
I inserted it with a text editor :smile:

VanessaE, 2018-02-23:
ok, fair enough

thinkyhead, 2018-02-23:
`M400` does a "synchronization" between moves, so it makes the previous move the last one in the buffer, and so it is the detached one.

That particular issue is in all versions of Marlin, btw.

VanessaE, 2018-02-23:
Ah, I understand.  Now, with `SEGMENT_LEVELED_MOVES` turned off (which *would* be appropriate here since my bed is nice and flat anyway), the stalling/skipping on X/Y that happens during short moves when in that mode is a showstopper. :frowning_face:

thinkyhead, 2018-02-23:
> Now, with `SEGMENT_LEVELED_MOVES` turned off … the stalling/skipping … that happens … when in that mode is a showstopper.

I'm having trouble parsing this. When `SEGMENT_LEVELED_MOVES` is off? Or "in that mode" (on)?

VanessaE, 2018-02-23:
Sorry, yes.  With `SEGMENT_LEVELED_MOVES` disabled/commented out, stalls/skips.

thinkyhead, 2018-02-23:
Hmm! That is weird, alright.

> switched back to @Sebastianv650's fork, and the skipping/stalling doesn't happen there

Well clearly we should compare them, try putting a few things the way they are in his branch, and see if any of those reversions fixes the issue.

VanessaE, 2018-02-23:
I just did.  I can reproduce the stalling in both branches by just commenting out that setting.

thinkyhead, 2018-02-23:
Oh, so it's just a setting. Poo.

VanessaE, 2018-02-23:
Yup.

thinkyhead, 2018-02-23:
So, stalling is noticeable during a normal G-code print, or just when doing synthetic moves?

VanessaE, 2018-02-23:
Noticeable (actually quite severe) during regular print moves e.g. the numbers at the end of the K-factor calibration pattern, but I can't be sure if it happens during...er... what do you mean by synthetic moves?

thinkyhead, 2018-02-23:
Typed in moves for testing purposes.

AnHardt, 2018-02-23:
@thinkyhead 
What we see in your test squares is a not joined last move/segment regardless of the `SEGMENT_LEVELED_MOVES` setting.
The difference is just the size of the last segment. When short, it is hard to distinguish it (acoustically) from the normal end of the line.

VanessaE, 2018-02-23:
Ah. Well, with this sequence, with the four G1 moves repeated several times in a row,
```gcode
G28
G29
G1 X100 Y100 F18000
G1 X101 Y100 F18000
G1 X101 Y101 F18000
G1 X100 Y101 F18000
```
... it doesn't cause a stall, whether I paste into the "send command" field or "print" it from a regular text file.

thinkyhead, 2018-02-23:
> What we see in your test squares is a not joined last move/segment regardless of the `SEGMENT_LEVELED_MOVES` setting. The difference is just the size of the last segment. When short, it is hard to distinguish it (acoustically) from the normal end of the line.

Yes, I'm way too aware of the details of that issue. Except how to fix it.

thinkyhead, 2018-02-23:
> It doesn't cause a stall, whether I paste into the "send command" field or "print" it from a regular text file.

Generally speaking stalls come from a starved planner when every move is a "final move." So chaining moves together prevents it from appearing. And we now make sure to chain the first move (another longstanding issue that we hope is fixed and not leading to side-effects).

It's too bad that we don't have debugging for these lowest-level issues that could be enabled at this time.

VanessaE, 2018-02-23:
When I say "stall" I don't mean the printer stops - I mean the motor skips a bunch of steps.

thinkyhead, 2018-02-23:
Well that's different. Motors may skip due to too many pulses per second, overheating drivers… Generally we also use the term "skip" to mean the motor tried to move, but hit too much friction and jumped backward a whole step. If that happens easily it can mean low current. Of course, you know all this…. The fact that changes to settings eliminates or reduces the issue is telling……

With hopes for a new release soon, I very much want to solve this. But I must sleep now, so I'll check in later today.

VanessaE, 2018-02-23:
No sleep for you! :smile:

thinkyhead, 2018-02-23:
I'm sleeping in a bowl of soup, so…

VanessaE, 2018-02-23:
I am suddenly reminded of the opening "scene" in Neuromancer (the old video game)

AnHardt, 2018-02-23:
@VanessaE 
Have you ever tried `#define ADVANCED_OK` (C*_adv.h)?
This will print the actual buffer states with every 'ok'.
I suppose a relation of a not well filled planner buffer with the stuttering you see.
To improve that you have to eiter print slower or to increase the segment sizes.
`SLOWDOWN` also may help.

Skipping steps in these low speed conditions let me assume a to high jerk. With some luck the steps won by breaking and lost in accelerating are equal and you don't see misdimensoned parts - but.

AnHardt, 2018-02-23:
@thinkyhead 
I'm well aware you are aware.
In a silent hour i'll check the 'optimizer' again. But already having played with it for days, i'm probably blind for the problem.

VanessaE, 2018-02-23:
I had not tried `ADVANCED_OK` before.  I've turned it on now, but I don't see any appreciable difference in the P and B values whether `SEGMENT...` is enabled or not, during the "problem" area of the test pattern.  Lots of "ok Nxxx P0 B3" during that time, in both cases.

As for jerk, I have X and Y set to 20, Z at 0.3, E at 5.0, as my machine performs well with those settings in any case.  For example, I printed this over the past day or so, using @Sebastianv650's LA 1.5 branch, with those jerk settings, `SEGMENT...` enabled, 0.1 mm layers, 0.6 mm line width, K=0.105 (for all colors, I haven't calibrated it per-color yet):

[image]
(high resolution image, click it, then click again for 100% size)

As you can see, it has well-defined edges and corners, and the original model file's faceting is clearly visible.  One of my better prints, if not for the fact that I still need to improve my layer cooling (you can see that on the edge of the side window facing the camera).  What you can't see is the subtle pattern visible on the back of the display stand, caused by the accel/decel of the segmented moves.

All of that said, "Error:Failed to enable Bed Leveling", yet it probed the bed just fine at print start. I've never seen this error before (or at least never noticed it).

AnHardt, 2018-02-23:
In @thinkyhead s test pattern we let run dry the buffer by purpose with the M400s.
In the LA pattern the buffer ideally should stay full, like in normal printing parts. It can fluctuate a bit but  shouldn't run dry or below 3.

VanessaE, 2018-02-24:
Here's something to try:  leave `SEGMENT_LEVELED_MOVES` enabled/un-commented, and set `LEVELED_SEGMENT_LENGTH` to something ridiculously high, e.g. 999.  When I do that, I get proper single-event moves without jitter or bisecting.  I'm not sure how well that'll work during a "real" print, but it works during the K-factor pattern, and no skipped X/Y steps during the pattern's numbers at the end either.

thinkyhead, 2018-02-24:
> leave SEGMENT_LEVELED_MOVES enabled/un-commented, and set LEVELED_SEGMENT_LENGTH to something ridiculously high, e.g. 999. 

Interesting. And, you only get leveling applied to the start and end of each G1 move.

VanessaE, 2018-02-24:
That seems to be the case, yes.

thinkyhead, 2018-02-24:
Try some experiments with your LCD entirely disabled… humor me, for curiosity's sake.

VanessaE, 2018-02-24:
My bot doesn't have an LCD.

thinkyhead, 2018-02-24:
Well f*** me then. I can't see what's being implicated here yet.

VanessaE, 2018-02-24:
Heh.  Sorry :smile:

thinkyhead, 2018-02-24:
At this point all I can think is to work backwards a week at a time and try points in time until the problem goes away, testing the worst-case scenario. And from this figure out what commit is contributing to this oddity.

VanessaE, 2018-02-24:
Are you not able to reproduce the issue using my configs?

thinkyhead, 2018-02-24:
I note that your `Configuration.h` is unchanged from the default, so maybe it's the wrong one…? And your `Configuration_adv.h` only has `LIN_ADVANCE` and `FWRETRACT` added to it, as far as effective settings go.

So, in short, I have not enabled `LIN_ADVANCE` and `FWRETRACT` and done some testing with the CR-10S yet. But I'm happy to take that on as my morning project tomorrow. And if you have other settings to pass on, I'll use those too.

VanessaE, 2018-02-24:
Strange, I thought I sent the right one.  Here they are again (and just to be sure, I unzipped the file to a work dir and double checked that the files I wanted were in there): [Configs-20180223-9c65890.zip](https://github.com/MarlinFirmware/Marlin/files/1753985/Configs-20180223-9c65890.zip)

VanessaE, 2018-02-28:
@thinkyhead:  I have still not yet seen any problems from leaving `LEVELED_SEGMENT_LENGTH` set to 999.  In fact, I had to remind myself that it's still set that way, because it seems to "just work". :smile:  I suppose any value larger than my printer's diagonal would have been fine, for that matter.

Sebastianv650, 2018-02-28:
Linking to #9494, @ikarisan noted:

>  Every time the nozzle reaches the location of the artifact the movement stops for approx. 0.5s and then continues at full speed.

There is definitly something going on regarding UBL, disabling segment splitting is a work around for this.  Would be nice if a leveling guru could have a look into this.

AnHardt, 2018-02-28:
Even with disabled extra `segment splitting` the moves will be split at the mesh lines. Else the mesh based levelings can't work.

thinkyhead, 2018-03-11:
> Would be nice if a leveling guru could have a look into this.

Ultimately it's got to be a planner issue, most likely related to changes in Z direction. You can try G-code that does small moves in Z in conjunction with XY moves and probably get the same effect.
```gcode
G28
M420 S0 ; no leveling
G1 Z10 F2400
G1 X10 Y10
G1 X20 Y20 Z0.1
G1 X30 Z-0.1
G1 X50 Z0.2
G1 X70 Z-0.2
```
If the G-code above generates strange pauses, that may give some clue what to look for. Long pauses like 0.5s could also indicate something weird, like the feedrate getting set very low at the planner level. It would be good to add some logging in the planner to uncover where it's going wrong.

Sebastianv650, 2018-03-11:
This should be fixed by #9914? @VanessaE I think I remember you mentioned it was fixed in another issue?

VanessaE, 2018-03-11:
@Sebastianv650, yes, it's fine now, at least for me it is.

closed 2018-03-11 by thinkyhead