MarlinFirmware/Marlin #9222: [Question] Most promising 32 Bit Platform for Marlin 2.x

mamama1, 2018-01-17:
Hi!

I was wondering what currently the most promising/best working 32 Bit platform for the 2.x Marlin branch would be?
Is it the Arduino Due? Some STM32?
What is the "reference platform" going to be?

Thanks!

mamama1, 2018-01-17:
ok thanks, so now the obvious question: why is avr green in that table and re-arm has an exclamation mark?

Roxy-3D, 2018-01-17:
Right now, there are two branches.  Bugfix-v1.1.x and Bugfix-v2.0.0.   They are almost identical from a feature set perspective.   But Bugfix-v2.0.0 is modified to have a hierarchical file layout.  That was needed to clean up the code and to support 32-bit. 

The AVR code is fully debugged and working in both branches.   The 32-bit code only exists (and works) in the Bugfix-v2.0.0 branch.   And while the code does work...   And some people are using it to print, it isn't entirely painless to get to that point.

mamama1, 2018-01-17:
i was stupid, sorry. of course avr is 8 bit.
i was wondering about SAM3X8E (due) too - why is it green and re-arm has an exclamation mark, even though it is the reference platform? which one works better currently?

thanks again

Roxy-3D, 2018-01-17:
Not many people are using Due hardware.  But they seem to think it is working well for them.   It maybe they don't know of problems that are exposed on that platform but they think it deserves a green color.

The Re-ARM board has a lot of people working with it and we know of a number of issues on that platform.   It is the reference platform, but it maybe a little bit behind Due right now.

mamama1, 2018-01-17:
i see, thanks!
do you have any idea when marlin 2.x will be approximately at a level like 1.1.x?

one year, two, four, ... any feeling?

Roxy-3D, 2018-01-18:
For AVR, it is basically there now.    For 32-bit platforms, I'm guessing 2 or 3 months.   Mostly, the big difference is the 32-bit platforms are using Platform-IO to build the code.   And that is not as easy to use as Arduino.    But Arduino now has an LPC-1768 add on so it maybe that we can bring that to life to make things easier.

mamama1, 2018-01-18:
very cool :-) i like the arduino environment so that'd be cool too :-)

p3p, 2018-01-18:
> But Arduino now has an LPC-1768 add on so it maybe that we can bring that to life to make things easier.

They have a functional lpc1768 core now?, I haven't seen it mentioned?

Roxy-3D, 2018-01-18:
I'm not sure it is 'they'.    There is an LPC-1768 core somewhere.   I'm not sure how solid it is.   But it is out there.   So it is possible we can use it.

closed 2018-02-11 by thinkyhead