MarlinFirmware/Marlin #9224: ABL still active after any motor has been powered off

MasterPIC, 2018-01-18:
On latest 2.0 bugfix (ABL) leveling stays active even if I power off a X, Y or Z motor.

tcm0116, 2018-01-18:
I don't really understand your issue. How do you know if leveling stays active when the motors are disabled? What negative effect are you seeing because of this?

MasterPIC, 2018-01-18:
> How do you know if leveling stays active when the motors are disabled? 

By Gcode command or by LCD display.

>What negative effect are you seeing because of this?

You can move by hand the nozzle and the printer won't update current coordinates. So when powering off X, Y or Z motor leveling should be deactivated.
Also without moving the nozzle by hand the nozzle can move by itself.

tcm0116, 2018-01-18:
>You can move by hand the nozzle and the printer won't update current coordinates. 

This is true for almost every printer out there. One would need to have encoders installed on the printer for it to be able to detect manual movement of the nozzle.

MasterPIC, 2018-01-18:
Keeping leveling active obliges to keep motors powered.

MasterPIC, 2018-01-18:
Maybe an option could be to reactivate leveling as soon as a new home command has been executed, with no need for recalibration...

thinkyhead, 2018-02-11:
Turning leveling off in response to motors turning off. Well, of course, if you move the carriage by hand, your next `G1` could led to bad things too. We're considering just forbidding any motion if the steppers have gone to sleep, until the machine is re-homed. That might be the best solution, ultimately.

tcm0116, 2018-02-12:
>We're considering just forbidding any motion if the steppers have gone to sleep, until the machine is re-homed.

Shouldn't `NO_MOTION_BEFORE_HOMING` cover that scenario?

AnHardt, 2018-02-12:
Currently we have two levels of not knowing where we are.   

* Before homing we have absolutely no clue about where we are. ('?')
* After disabling a stepper we *may* have a little shift by having lost a view steps (gravity, springiness, ...) (' ')  It's just mend as a warning. Lots of machines disable steppers/drivers to keep them cool without any negative effect. For that you can switch off the warning.
Moving any axis by hand during the stepper is disabled is not covered/mend, That always needs rehoming.

Automatic rehoming is not a thing i want to see when a part could be on the bed (always?)
If homing has to be done manually it's no real additional task to reenable leveling (host macros).

tcm0116, 2018-02-12:
What if homing was required after the steppers have been disabled and there
is not an ongoing print? On my machine, it's configured to home
automatically at the start of M600 if the machine has not been homed.
However, sometimes a print will finish, and the steppers will be enabled,
and I end up moving the bed around removing the part. I'll then run M600 to
change filament before starting the next print, but it does not home
because it's only in the warning state and not the unhomed state.

thinkyhead, 2018-02-20:
I agree, this would be a good option to add. With `SLEEP_REQUIRES_REHOME` enabled the machine will chirp at you on move commands until it has been homed. Maybe it should be on a per-axis basis. If Z hasn't gone to sleep, you can still move Z, etc.

boelle, 2019-02-20:
@thinkyhead maybe add the feature request label ?

boelle, 2019-06-22:
missing label