MarlinFirmware/Marlin #9225: [FR] Ability to cool instead of heat with output

nicksears, 2018-01-18:
I'm working on a project where I need to cool a platform instead of heat it. I see nowhere in the code to enable this so I've come up with a workaround:

Use a logic inverting chip (https://www.ebay.com/itm/400083743600) and use output pin 4/5/6/11 instead of 8/9/10 for extruder or bed heater output (I'm using 4 instead of 8) for the bed.

The chip takes 5v and ground, as well as the input signal and outputs the opposite of that logical signal (http://www.ti.com/lit/ds/symlink/sn74ahc1g04.pdf).

This way when the temperature is too cold (usually turns on) it is off (not cooling). When the temperature is too hot (usually turns off) the the output is on (cooling). This may be more easily implemented with a mechanical relay (for the bed at least if there's a low switching frequency).

The signal out can then control an SSR or for now I'm using a IRF520 breakout board (https://www.ebay.com/itm/232518999508).

This is the quick and dirty way to do it and by default it is ON, so I need an extra switch (mechanical relay) controlled with an extra pin, that I turn on when I want to engage this cooling circuit. It would definitely be better to have an option to just cool INSTEAD of heat, or it would be great to have competing outputs (2-stage) like found in some PID systems (https://www.amazon.com/dp/B01MDSWXY4).

thinkyhead, 2018-02-11:
You will have to modify Marlin yourself, rewriting the heating routines to look for drops in temperature instead of rises. Possibly the PID code will not be as usable because it is written with the assumption of rising temperatures. In any case, if you lack the skill to make these changes, the best thing you can do is collect $$$ and hire a developer to do this for you.

closed 2018-02-11 by thinkyhead

nicksears, 2018-02-14:
Acknowledged. For anyone wanting to attempt this, I was able to get it to work pretty well with my logic inverter and a relay to switch between heating and cooling circuits. 

The one thing it's missing is a PID_MIN, is this possible/enabled in any firmware? Since we're using the opposite signal we want it to always be on at a minimum rate (which is the PID_MAX when inverted). 

If not, would this be easy to enact in the firmware? I've done some significant firmware modification and written new code before, but unless this an existing concept, it might be hard to modify the existing code to make this happen. I think this could be useful in other situations, though, not just my special case...