MarlinFirmware/Marlin #9228: [1.1.8] G26 validation grid is too small?

hoppersoft, 2018-01-18:
I just upgraded to 1.1.8. I have set the bed dimensions to 200x200 with my fixed z-probe set to x +40 and y -48 and a UBL inset of 2. When I perform a G26 validation grid, the grid is inset very far into the bed area (1.1.6 would have the grid stretch to the edges of the bed - inset distance). Is this a bug, or have I screwed something up in my config (attached)?

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1641657/Configuration.zip)

Roxy-3D, 2018-01-18:
Did you just download 1.1.8 today?   I ran into a similar issue and changed it for my Folger Tech i3-2020 (which has a 200mm x 200mm bed also) with this:   https://github.com/MarlinFirmware/Marlin/commit/80c1afde0f9c6e810da326df2244af52408ea3dd#diff-ffba036d1fad5d16fc1b010038edf9e0

hoppersoft, 2018-01-18:
I pulled it Monday afternoon.

Yeah, those changes look like they'll do the trick. I'll give it a shot. Thanks!

Roxy-3D, 2018-01-18:
Also...   There are other reasons you want to download the full branch and not just add those changes to what you have.   There are other nice things added to the G29 and G26 code here:   https://github.com/MarlinFirmware/Marlin/pull/9227

hoppersoft, 2018-01-18:
Thanks for the tip! Has that been merged back down to 1.1.8, or should I just cut over to the bugfix branch?

hoppersoft, 2018-01-18:
Disregard: just realized 1.1.8 is a tag, not a branch.

Roxy-3D, 2018-01-18:
Bugfix is always ahead of the base branch.   You want to use https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x       I'll be moving those changes to bugfix-v2.0.0 soon, but they are not there yet.

hoppersoft, 2018-01-19:
Well, nuts. I cut over to bugfix, and now I'm getting
` static assertion failed: UBL_PROBE_PT_1_Y can't be reached by the Z probe`

I checked Conditionals_post, and the math looks right; the probe is in the negative direction, but the #define for MIN_Y_PROBE uses the maximum of the minimum Y coordinate (0) and the minimum hotend position + sensor offset (-48), which would be 0 and is within the range of bed Y coordinates 0-200.

(Same configuration as above, btw)

Roxy-3D, 2018-01-19:
That shouldn't be a big deal...   Just make the UBL_PROBE_PT_1_Y point on the bed.   Most likely you won't even see a need to do mesh tilting.   And certainly not 3-Point mesh tilting.    Just shift the UBL_PROBE_PT_1_Y such that the compiler doesn't complain.

hoppersoft, 2018-01-19:
Oh, dear God. I'm just very bad at math and miscalculated the reachable area. Please disregard and act like I never said anything.

closed 2018-02-14 by thinkyhead