MarlinFirmware/Marlin #9229: [bugfix-1.1.x] M303 Heating failed

Xcorpia07, 2018-01-18:
Hello everybody, sorry for my poor English. It's my first time in this website. 

I'm using a Anet A6 and decide to change the firmware to Marlin. I probe the ultimate stable version 1.1.8 and ours bugfixs.

### Bug Report
- Description: ---M303 Heating failed
- Expected behaviour: PID Autotune information  
- Actual behaviour: Error Heating failed 
- Steps to reproduce:
  - Do this: Init Anet A6 with Marlin 1.1.x
  - Do that: Send commando M303 E0 S200 C8 or M303 E-1 S50 C4

I use Configuration.h` and `Configuration_adv.h examples are in the ultimate bug-fix

The problem is only in the M303 PID autotune parameter.
Since with preheated or with M104, M109 or M140, M190 I have no problem.
That's why I think it's not due to any parameter: 

```
#if ENABLED(THERMAL_PROTECTION_HOTENDS)
  #define THERMAL_PROTECTION_PERIOD 60       // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 10     // Degrees Celsius

  #define WATCH_TEMP_PERIOD 60                // Seconds
  #define WATCH_TEMP_INCREASE 5               // Degrees Celsius
#endif

#if ENABLED(THERMAL_PROTECTION_BED)
  #define THERMAL_PROTECTION_BED_PERIOD 60    // Seconds
  #define THERMAL_PROTECTION_BED_HYSTERESIS 5 // Degrees Celsius

  #define WATCH_BED_TEMP_PERIOD 180                // Seconds
  #define WATCH_BED_TEMP_INCREASE 2               // Degrees Celsius
#endif
```

I would also like to know which is the most appropriate method for PID Autotune, if it is from room temperature or preheat previously.
Because if I preheat the hotend or the hotbed I do not have that problem either.

TGMods, 2018-01-18:
I have the same problem on Bugfix 2.0.
I had a heater problem and I had to rebuild my hot end. I also took the opportunity to replace my thermistor which was giving me problems.
On Tuesday afternoon (UK time) I uploaded Bugfix 2.0 on my Sainsmart 2-in-1. Once everything looked good I tried to run M303 with the same error output as in the Repetier Server Report.txt.
The only older copy of Marlin I had available that I was sure did not have a problem with PID tuning was Marlin 1.1.4. I built and uploaded that version which ran the PID tuning correctly. I then downloaded and built the latest Bugfix 2.0. I did not have the time yesterday to try PID tuning.
I finally had time today and it failed.
My configuration.h files from 1.1.4 and Bugfix 2.0 along with the Repetier Server log are enclosed in the attaced zip file.

[Configs_and_Error_Report.zip](https://github.com/MarlinFirmware/Marlin/files/1643487/Configs_and_Error_Report.zip)

Xcorpia07, 2018-01-18:
I do not understand the changes. Now I downloaded the last bugfix.1.1.x 1/17/18 20:51 date of the files.

The problem changed a bit. Now I can use M303 E0 S200 C8, the hotend can heat  to 200º but in the third cycle, hotend don't heated more and get the error killed Heating failed. 

And the hotbed I send M303 E-1 S50 C4 don't heat anything and last time get error Thermal Runaway. 

I use the same Configuration.h file that the order question. I don't see the changes in the code that affect to this changes in the problem.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1643734/Configuration.zip)

Thanks for yours time and help.

riisefrutti, 2018-01-18:
I have a sort of same issue, which I cant wrap my head around. PID Autotune fails after a few seconds, psu goes to rest. Then Heating failed, and the printer halts.  To attempt to localize the fault I ran preheat PLA/ABS. Found that they heated to set temp, and held them in the normal range. Tried to upload new firmware, issue is the same. Also tried running PID autotemp several times, seems to run in segments of 30-40 deg C before cutting power to hotend.

Xcorpia07, 2018-01-19:
# [Bugfix 2.0] M303 Pid tuning failure https://github.com/MarlinFirmware/Marlin/issues/9233#issuecomment-358811688

Xcorpia07, 2018-01-19:
Hello Now I can continue flashing another one. I downloaded the lasted bugfix 1.1x 19-01-18 and I tried PID Hotend and HotBed and the problem persist. 

`echo:SD card ok
>>> M303 E0 S200 C8
SENDING:M303 E0 S200 C8
PID Autotune start
Error:Heating failed, system stopped! Heater_ID: 0
[ERROR] Error:Heating failed, system stopped! Heater_ID: 0

Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!`

The temperature go up to 180ª and stop heating, I see the led in the board, and a few seconds the Pronterface and Screen say the error. 

If I Preheat HotEnd and HotBed no error healting.

But if I preheat the HotEnd I have this error 

```
>>> M303 E0 S200 C8
SENDING:M303 E0 S200 C8
PID Autotune start
 bias: 56 d: 56 min: 198.42 max: 206.43
 bias: 56 d: 56 min: 198.42 max: 202.28 Ku: 36.98 Tu: 29.82
 Classic PID
 Kp: 22.19 Ki: 1.49 Kd: 82.71
Error:Thermal Runaway, system stopped! Heater_ID: 0
[ERROR] Error:Thermal Runaway, system stopped! Heater_ID: 0

Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!
```

And Now If I preheat the HotBed, I can tried PID. But I no sure if this parameters are OK, because I tested with the bed  preheated.

GMagician, 2018-01-21:
Since 2.0 has been fixed and same fix has been merged for 1.1 I think this can be closed

riisefrutti, 2018-01-21:
Tried and found successful on 1.1. Thank you! 

Sendt fra min iPhone

Xcorpia07, 2018-01-21:
Thanks the fix its ok. I close

closed 2018-01-22 by thinkyhead