MarlinFirmware/Marlin #9233: [Bugfix 2.0] M303 Pid tuning failure

TGMods, 2018-01-18:
### Bug Report
- Description: M303 PID Auto tune failure
- Expected behaviour: Complete Auto Tune resuling in good Kp, Ki, Kd, values.
- Actual behaviour: Heat failure at around 140C
- Steps to reproduce:
  - M303 E0 S198 C8

I then built and uploaded Marlin 1.1.4 and tried the above steps again. This was successful can gave me a valid PID.
Rebuilding and uploading Bugfix 2.0 resulted in a PID failure again. Configs from 1.1.4 and Bugfix 2.0 attached in the zip file. Also attached in zip file is the failing Repetier Server log.

[Configs_and_Error_Report.zip](https://github.com/MarlinFirmware/Marlin/files/1644389/Configs_and_Error_Report.zip)

p3p, 2018-01-18:
Have you tested bugfix-1.1.x, or is this just a 2.0 issue?

revilor, 2018-01-18:
From the Repetier output it looks like heating the hotend stops at 150°. This is the value of BED_MAXTEMP which probably is no coincidence. MIght also be related to #9223

TGMods, 2018-01-18:
I haven't tried bugfix-1.1.x but I know @Xcorpia07 was having the same problem with Bugfix-1.1.x ([#9229](https://github.com/MarlinFirmware/Marlin/issues/9229))

Xcorpia07, 2018-01-18:
I haven't tried bugfix 2.0.x because I tried to put and the lcd screen is not ok on Anet A6.

And now I have the bootloader of Aner board corrupt because I tried to restore the board with arvdude. I recover the firmware and worked but i can't flash other firmware. 

Tomorrow I tried to Burn bootloader with Arduino Mega for ISP and continue with the investigation of the problem.

marcalcc, 2018-01-19:
I wanted to open an issue regarding the PID autotune on 2.0 Bugfix as well.

On my case:

Marlin 2.0 Bugfix (downloaded on January the 18th, 2018)
MKS Sbase 1.3 Board

I first tried to autotune the Hotend and failed (I didn't notice the Temperature). I reset and tried again. The autotune finished as expected (and I got the values) but the system became unresponsive after one or two minutes.

Next, I tried to PID autotune the heated bed to no avail. Between 30 and 35 degrees, the system becomes unresponsive too.

I have the feeling it is a software related issue as I tried Smoothieware before Marlin 2.0 Bugfix and I was able to PID autotune both the Hotend and the heated bed without problems.

Thanks

Xcorpia07, 2018-01-19:
I added a new coment.
**[bugfix-1.1.x] M303 Heating failed https://github.com/MarlinFirmware/Marlin/issues/9229#issuecomment-359092274**

GMagician, 2018-01-20:
I made some test. I noticed that output is turned off too soon.
'heated' status was not reached, current temp was 172.89 target 200 watch temp to raise error 174.2 and output off...continuing

GMagician, 2018-01-20:
ok I saw that when output is disabled the variable soft_pwm_amount[0] is set to 0.
I also noticed that sometimes this variable is set to 0 and then 127 again.
Can't find where is the real issue

GMagician, 2018-01-20:
Ok,  found! It's thermal manager that turn off output. There is a call to 'safe_delay' somewhere that trigger the fault...

Xcorpia07, 2018-01-20:
But only affect in PID Autotune. The thermal manager is for general control of temperature??

GMagician, 2018-01-20:
Yes, but during pid is called 'print_heater_state' that call safe_delay that call thermal manager and is call, sometimes, stop output

GMagician, 2018-01-20:
That's the guilty code:
      // Report heater states every 2 seconds
      if (ELAPSED(ms, next_temp_ms)) {
        #if HAS_TEMP_HOTEND || HAS_TEMP_BED
          print_heaterstates();
          SERIAL_EOL();
        #endif
in pid_autotune

TGMods, 2018-01-21:
@GMagician Thanks PID tuning works the way it should with the latest Bugfix 2.0

closed 2018-01-21 by TGMods