MarlinFirmware/Marlin #9234: Compiling Bugfix-2.0 for DUE error

maar1201gh, 2018-01-18:
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\HAL_timers_Due.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\Servo_Due.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\persistent_store_impl.o
Compiling .pioenvs\DUE\src\src\HAL\HAL_DUE\u8g_com_HAL_DUE_st7920_sw_spi.o
Marlin\src\HAL\HAL_DUE\HAL_timers_Due.cpp: In function 'void HAL_timer_interrupt_enabled(uint8_t)':
Marlin\src\HAL\HAL_DUE\HAL_timers_Due.cpp:124:57: error: new declaration 'void HAL_timer_interrupt_enabled(uint8_t)'
void HAL_timer_interrupt_enabled(const uint8_t timer_num) {
^
In file included from Marlin\src\HAL\HAL_DUE\../HAL_DUE/HAL_Due.h:38:0,
from Marlin\src\HAL\HAL_DUE\../HAL.h:40,
from Marlin\src\HAL\HAL_DUE\HAL_timers_Due.cpp:35:
Marlin\src\HAL\HAL_DUE\../HAL_DUE/HAL_timers_Due.h:116:6: error: ambiguates old declaration 'bool HAL_timer_interrupt_enabled(uint8_t)'

cjsoong, 2018-01-19:
void HAL_timer_interrupt_enabled(const uint8_t timer_num) {
  const tTimerConfig * const pConfig = &TimerConfig[timer_num];
  return pConfig->pTimerRegs->TC_CHANNEL[pConfig->channel].TC_IER == TC_IER_CPCS;
}
change to 
bool HAL_timer_interrupt_enabled(const uint8_t timer_num) {
  const tTimerConfig * const pConfig = &TimerConfig[timer_num];
  return pConfig->pTimerRegs->TC_CHANNEL[pConfig->channel].TC_IER == TC_IER_CPCS;
}

Roxy-3D, 2018-01-19:
I changed it in the bugfix-v2.0.0 branch.   Hopefully you are correct!

closed 2018-01-21 by maar1201gh