MarlinFirmware/Marlin #9238: [2.0 Bugfix] Digipot I2C Motor current order

marcalcc, 2018-01-19:
### Bug Report
- Description: Setting the current on I2C digipots does not follow assigned motor pins
- Expected behavior: follow assigned pins
- Actual behavior: not following assigned pins
- Steps to reproduce: try out different configurations

I'm testing Marlin 2.0 Bugfix on a MKS Sbase 1.3  Unfortunately my boar came with a damaged Z-Motor driver. Since I don't need double extrusion, I plugged the Z motors (Prusa i3 clone -type printer) on the E1 socket. To make it work I rearranged the pin configuration on the pins_MKS_SBASE.h file and assigned the E1 socket pins to the Z motor. So far, so good. I got it to work.

However, as I was fine-tuning the current on Configuration_adv.h, on the line:

#define DIGIPOT_I2C_MOTOR_CURRENTS { 1.2, 1.2, 1.2, 1.2, 0.2 }

I was trying to adjust the Z motor current by changing the third value inside the brackets, before realizing that I should be changing the fifth value instead, meaning that the order inside the brackets follows the layout of the board instead of the X, Y, Z order.

While I'm not saying it's a bug in the strict sense of the word as I can see the logic behind the way it is as well, I would suggest that this could be clarified by a commented-out annotation somewhere. It was a bit confusing in the beginning...

Thanks a lot

thinkyhead, 2018-02-14:
Please test with the latest `bugfix-2.0.x` branch to see if we fixed this issue. If the problem has been solved then we can close it. If you still see the bad behavior we should investigate further.

thinkyhead, 2018-02-14:
Ah, I see your point. I'll add commentary to the configuration to make this clear.

closed 2018-02-14 by thinkyhead