MarlinFirmware/Marlin #9239: [FR] feature question re hotbed

ruggb, 2018-01-19:
I am updating my printer with a 120V hotbed. It has no "on" indicator. I would like to add one that comes on when the bed temp is over about 40°C as opposed to just being on when the bed is powered. 
Is that available? Can I add it easily? Can you add it.
Seems like a good safety feature to have.

ruggb, 2018-01-23:
I found the function but it appears to be dependent on the defined LCD controller. Can I just move those pin defines directly under Ramps and set them for unused pins?

thinkyhead, 2018-02-14:
See:
```cpp
/**
 * Printer Event LEDs
 *
 * During printing, the LEDs will reflect the printer status:
 *
 *  - Gradually change from blue to violet as the heated bed gets to target temp
 *  - Gradually change from violet to red as the hotend gets to temperature
 *  - Change to white to illuminate work surface
 *  - Change to green once print has finished
 *  - Turn off after the print has finished and the user has pushed a button
 */
#if ENABLED(BLINKM) || ENABLED(RGB_LED) || ENABLED(RGBW_LED) || ENABLED(PCA9632) || ENABLED(NEOPIXEL_LED)
  #define PRINTER_EVENT_LEDS
#endif
```

closed 2018-02-14 by thinkyhead