MarlinFirmware/Marlin #9241: [FR] [1.1.x] Relax thermal protection for a minute after M106 when part cooling fan starts

ejalal, 2018-01-19:
Hi Guys,

I use my printer in the garage and it's cold nowadays and I'm having an issue with the thermal protection stopping the print when the part cooling fan starts spinning after the first layer.
The hotend recovers the initial temperature very quickly, but the heatbed loses 2°C very quickly and recovers after a few minutes.

Is it possible to introduce a new feature to relax the thermal protection for a couple of minutes (number of minutes could be configurable) when Malin receives a M106 to start the part cooling fan ?

Thanks for the great work :+1:

ikarisan, 2018-01-19:
I like the idea. Sometimes I have the same problem with my nozzle and parts where air accumulates. Then the nozzles temperature goes down from 210°C to < 200°C for too much seconds and the printer stops. So I had to make a special Simplify3D profile for printing these parts. Every layer I am increasing the fan percentage by 10-20%. Depending on the model, it then takes up to 10 layers until the fan reaches 100%.

revilor, 2018-01-20:
I don't know what printer and hotend you guys are using but too me it looks like the solution to your issues would be a heater cartridge with more power. I just replaced the original 30W cartridge of my E3D Volcano with a 40W one. Before it was not possible to run PID autotune with the part cooling fan at 100% but with the 40W cartridge the hotend holds the target temperature perfectly no matter the status of the fan.

ejalal, 2018-01-22:
I already use a 40w cartridge for my hotend with an e3d silicone sock, the issue is with the hotbed that takes time to recover the temp loss (2 to 4°C).
The best solution is to do as @ikarisan said, slowly increase the fan speed from something like 50% on the second layer to 100% by 10-20% increments per layer.

revilor, 2018-01-22:
As you have a 40W heater already that's not the issue. Which leaves you with two possible reasons for your problem:

- maybe the part cooling fan is too strong. You can try to reduce the maximal fan speed in your slicer settings
- your settings for THERMAL_PROTECTION_PERIOD and THERMAL_PROTECTION_HYSTERESIS in Configuration_adv.h might be too low. You can try to increase hysteresis or watch period to give the hotend more time to recover before thermal protection triggers.

closed 2018-02-14 by thinkyhead